Ultimaker/Cura #16142: CFFFP_skull03-sup_fixed

FredericJULIEN, 2023-07-14:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Anycubic Kobra Neo

### Describe your model

No

### Add your .zip here ⬇️

[CFFFP_skull03-sup_fixed.zip](https://github.com/Ultimaker/Cura/files/12048294/CFFFP_skull03-sup_fixed.zip)

GregValiant, 2023-07-14:
Thanks for the report and the file.
I didn't find any errors in the model.  It simply won't slice in Cura 5.4.0.  I was able to slice it in 4.13.1 and PrusaSlicer.
Even after simplifying the model in 3D Builder it still would not slice.
The Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic