Ultimaker/Cura #16149: Print one at a time order

ganzey, 2023-07-14:
### Is your feature request related to a problem?

The feature print one at a time is pretty much useless when you cannot pick what is printed in what order, and it seems 100% random how it picks what is printed in which order as it is. 

### Describe the solution you'd like

add that feture

### Describe alternatives you've considered

other slicers that do have this

### Affected users and/or printers

everyone

rspangit, 2023-12-06:
This would be particularly useful when some parts are more likely to fail than others.  If I'm running an overnight job, I could print the fail-prone parts last, so that I'd at least get the good prints before the bad one stick to the head and drag back and forth across the print bed.

GregValiant, 2025-01-15:
This feature has been added in current versions of Cura.
Can this be closed?
[image]

closed 2025-01-29 by github-actions[bot]