Ultimaker/Cura #16151: Brim working only in some parts

RenatoMaterializa, 2023-07-14:
### Cura Version

5.4.0

### Operating System

Windows 10 64bit

### Printer

Creatbot D600 PRO

### Reproduction steps

Open file and slice
Change line width to see brim changing parts
Brim only on outside works fine

[image]

### Actual results

The brim works only in some parts, and it's random witch part the brim will work,
If I change the line width the brim change parts
the brim works completely only in a few parts
on the other parts there is only 1 or 2 lines of brim
(I also tried to use a custom support of 0.3mm to force a brim but it didn't worked)

Line width of 0.6mm:
[image]

Line width of 0,7mm:
[image]

### Expected results

all the parts with brim on the inside and on the outside

Same .3mf project on CURA 5.3.0
[image]

### Add your .zip and screenshots here ⬇️

[Brim not working.zip](https://github.com/Ultimaker/Cura/files/12053835/Brim.not.working.zip)

Cura 5.4.0:
[image]

With Cura 5.3.0:
[image]

GregValiant, 2023-07-14:
Thanks for the report.
I can reproduce this problem.
The Cura team will take a look.

@MariMakes this is one I haven't seen before.
[image]

jxer, 2023-07-15:
I also have this issue with support brims, they are not working at all on the inside of the supports, and they don't start till layer 2

5.1.1 doesn't have this issue, no setting adjustments fix this

jxer, 2023-07-15:
5.4 https://i.imgur.com/HgszpJh.png
5.1.1  https://i.imgur.com/pZOCIkk.png

jxer, 2023-07-15:
5.4 layer 2 first layer of support nothing at layer 1

https://i.imgur.com/39xBG6Y.png

MariMakes, 2023-07-17:
Hey @jxer,

Thanks for your report 👍
Your inconsistent brim is the first time I've seen this 🤔 
I also noticed that the brim changes based on the amount of objects on the buildplate.
That makes me think it might be a rounding issue.
[image]

I'll bring it up to the team to see what they can do to improve it. 
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞 

Your issue with the support brim missing, reminds me of: https://github.com/Ultimaker/Cura/issues/14983
It's more helpful to us to log separate issues in separate tickets, that way we can better keep track of how they are resolved.

RenatoMaterializa, 2023-07-17:
Hi @MariMakes , thanks for the reply,

I was testing a little more and found out that in one of the files that I printed, that rotating the part fixed the problem,

I printed two parts to see if everything was working before increasing the amount of parts in the buildplate, 

so I went to check what I did to fix it and It looks like rotating solved.

I don't remember if I sent this file with only two parts but here is the link to download

https://drive.google.com/file/d/1OF9p7nA1oXR--VI9tUGSsi4ERCvhfsa0/view

https://github.com/Ultimaker/Cura/assets/128747731/c3706be2-933c-425c-9cce-6bd72c6390a8

MariMakes, 2023-07-21:
Hey @RenatoMaterializa,

Thanks for the update, I'm glad you are not stuck anymore. 😊 

I also have an update on our side, when I showed your issue to the team, they noted that it should be resolved by fixing another bug they were working on here: https://github.com/Ultimaker/CuraEngine/commit/2ffffac40e8bee8a0dde973d316b4e0839834acc
[image]

That means that you can expect a fix in the next stable release. 
Thanks for the report and the great test case 👍

closed 2023-07-21 by MariMakes