Ultimaker/Cura #16152: Slincing failed Flsun v400 cura 5.4.0

chabz2008, 2023-07-14:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Flsun v400

### Name abnormal settings

Support settings

### Describe model location

In the center

### Describe your model

No

### Add your .zip here ⬇️

[ezyzip.zip](https://github.com/Ultimaker/Cura/files/12059151/ezyzip.zip)

GregValiant, 2023-07-16:
Thanks for the report and the file.
There is nothing wrong with your model but it won't slice laying down.  I did get it to slice when it's vertical, but that isn't functionally acceptable.
I'll mark this as a duplicate and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic