Ultimaker/Cura #16156: Dragon Pencil cup CRAP

meier2009, 2023-07-15:
### Cura Version

5.4

### Operating System

Win 11

### Printer

Artilery X1

### Name abnormal settings

Standard

### Describe model location

Center and all around

### Describe your model

YOUR Dragon Pencil 

### Add your .zip here ⬇️

EVEN the Model itself is NOT watertight and has errors in Original
next you cant Slice it.
Error

No repair worked , nothing!

GregValiant, 2023-07-15:
Thanks for the report.
You are correct, there is a model error in the base that can keep it from slicing.

@MariMakes here is another one.

MariMakes, 2023-07-17:
Hey @meier2009,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
It could be that your settings are different than people who are able to succesfully slice and print it. 

To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-08-01 by github-actions[bot]