Ultimaker/Cura #16157: Simple Tree and Slicing Failed with Tree Supports

sarahcath, 2023-07-15:
### Cura Version

5.4.1

### Operating System

Windows 11, Version 22H2, Build 22621.1992

### Printer

Ultimaker 2+

### Name abnormal settings

It only crashed when I selected the Tree Supports

### Describe model location

Does not fail when rotated.

### Describe your model

No, No, and No.

### Add your .zip here ⬇️

[Slicing Failed.zip](https://github.com/Ultimaker/Cura/files/12061643/Slicing.Failed.zip)

GregValiant, 2023-07-15:
Thanks for the report and the file.
This sliced for me once I set the Initial Layer Height to .20.
This bug has been noted before so I'll mark this as a duplicate.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic