Ultimaker/Cura #16158: Incomplete first layer

kdale2006, 2023-07-15:
### Cura Version

5.4.0

### Operating System

Windows11

### Printer

Creality CR-10 Smart Pro

### Reproduction steps

Open STL file
Slice
Save file
Save project 
Print
[image]

### Actual results

I have a part (Bottle Opener) that has a bunch of small fingers around a radius.
Everything prints on the first layer except the fingers.
The fingers do print on the second layer but won't stick to the bed because of the first layer space

### Expected results

I expect the complete first layer to print
[image]
[image]

### Add your .zip and screenshots here ⬇️

[CR10-29_0-V6-New Bottle Open.zip](https://github.com/Ultimaker/Cura/files/12062040/CR10-29_0-V6-New.Bottle.Open.zip)
[image]
[image]
[image]

kdale2006, 2023-07-15:
I did not have this issue with the previous version of Cura.
I am using the same STL file

GregValiant, 2023-07-15:
Thanks for the report.
This looks to be a settings problem.
Make sure "Initial Layer Horizontal Expansion" and "Hole Horizontal Expansion" are set to 0.0.
You could also enable "Remove all holes" and see if it gets rid of the lettering.

closed 2023-07-15 by GregValiant

kdale2006, 2023-07-16:
Thanks for the reply and things to try.
Making the "Initial Layer Horizontal Expansion" 0.0 solves the problem.  It
looks like even -0.1 causes some issues and -0.2 makes the fingers
disappear.  I was using -0.3 to take care of some "Elephant Foot" issues.
Has something new been added to take care of the "Elephant Foot '' problem
that pops its ugly head up.
Ken