Ultimaker/Cura #16162: First layer Tree Support Issue

omrfrkmll, 2023-07-15:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 V2

### Reproduction steps

1. load the object
2. slice it
3. go to preview section
4. do to first layer
5. --!! there is no tree support first layer

### Actual results

tree support is not there

### Expected results

tree support generated in first layer

### Add your .zip and screenshots here ⬇️

[image]
as you can see here there is little circles but in next image as second layer bigger tree supports comes.
[image]

this error not seen in 5.2.2 version of cura slicer

[cura.zip](https://github.com/Ultimaker/Cura/files/12062641/cura.zip)

GregValiant, 2023-07-15:
Thanks for the report.
This is a known bug that occurs when Bed Adhesion is set to "None" and "Support Brim" is turned on.  Turn on a "Skirt" and it should be OK.
I'll mark this as a duplicate.  Let us know if that workaround helps.

jxer, 2023-07-15:
> Thanks for the report. This is a known bug that occurs when Bed Adhesion is set to "None" and "Support Brim" is turned on. Turn on a "Skirt" and it should be OK. I'll mark this as a duplicate. Let us know if that workaround helps.

Please tell me this is a priority fix, using a skirt breaks a majority of my models in my business as I use the ENTIRE build plate for printing and have never used Adhesion

GregValiant, 2023-07-16:
If the skirt fixes the problem then you can try setting the Skirt Distance to 1mm and the number of loops to 1.  A new feature of skirt/brim is that the model can now go right to the edge of the build plate and the skirt/brim will stop at the edge.
[image]

jxer, 2023-07-16:
> If the skirt fixes the problem then you can try setting the Skirt Distance to 1mm and the number of loops to 1. A new feature of skirt/brim is that the model can now go right to the edge of the build plate and the skirt/brim will stop at the edge. [image]

i understand but is there a plan to fix this soon, this breaks automation platforms as we can't clear skirts.

GregValiant, 2023-07-16:
I think you would need to use a previous version of Cura.  I don't work for UltiMaker but from what I've seen they do about 4 releases a year and 5.4.0 just cam out.  It will be at least a few months.

@nallath is this bug on the hit-list?