Ultimaker/Cura #16164: slicing error

DaveWeimer78, 2023-07-16:
### Cura Version

5.4.0

### Operating System

windows 10 pro

### Printer

custom/ ender3 s1 pro

### Name abnormal settings

no

### Describe model location

no

### Describe your model

yes

### Add your .zip here ⬇️

[CFFFP_CrystalDragon_NewLegs_Curled.zip](https://github.com/Ultimaker/Cura/files/12063127/CFFFP_CrystalDragon_NewLegs_Curled.zip)

GregValiant, 2023-07-16:
Thanks for the report.
The Crystal Dragon has been here several times and there are different versions out there.  Some slice with no problem, but the models with errors just don't slice.
This is your model in Mesh Mixer.  The magenta and red balls indicate errors in the model.
[image]

In addition you are asking for a .30 Initial Layer Height, 0.20 Layer Height and .04 Line Width from a 0.2 nozzle.
Here is the repaired model sliced with a 0.4 nozzle, 
[image]

I'll leave the bug label on this but between the settings and the bad model trying to chase down a bug will be pretty much impossible.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic