Ultimaker/Cura #16165: Wall printing not optimized with "Optimize wall printing order" Cura version 5.4.0

kurolob, 2023-07-16:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Custom

### Reproduction steps

Load a stl file that contains a cube with 3 vertical holes through the cube. In cura 5.4.0 and 4.10.0
Select "Optimize wall printing order" in every version
4.10.0 prints the inner wall and then outer wall for the first hole. Then inner wall and outer wall for the second hole. Then inner wall and outer wall for the third hole. Then inner wall and outer wall for the cube.
5.4.0 prints the inner wall for every wall. Then outer wall for every wall.
This can be a issue when printing a model with many islands for example.
[image]
[image]

### Actual results

The wall are not optimized that may results in problems when printing "islands".

### Expected results

Should print like 4.10.0

### Add your .zip and screenshots here ⬇️

[image]
[image]
[porta bateria com carregador e conversor-Body003.zip](https://github.com/Ultimaker/Cura/files/12063161/porta.bateria.com.carregador.e.conversor-Body003.zip)

GregValiant, 2023-07-16:
Thanks for the report.
The Cura team will take a look.
It appears to me that it has something to do with the Wall Line Count.  Setting that to 3 produces a slightly different toolpath.  It also appears that the issue would be with the Inner-Walls.

sityware, 2023-11-07:
I think I have the same issue with holes, on the bottom layer it causes so much back and forth I'm having unnecessary whisper strings.

It first draws the middle wall for each hole, then comes back to draw the inner hole of each, and then finally the outer wall for each.

[image]

GregValiant, 2023-11-08:
I get the same thing in my prints.  Rather than staying over one hole/feature and finishing it, Cura wants to do all the extrusions strictly by line type.  All the Inner-Walls, then all the Outer-Walls, then all the Extra Skin Walls, then the skin.  Certain combing modes along with monotonic can make the travel time pretty extensive.

sityware, 2023-11-08:
I've created an example STL here that demonstrates, you can download here.
[https://mega.nz/file/8bxXTbIA#X2AQNHA6484EBXmXUEsFqz2OZB-vUeE5HVToRgEa4Io](url)

[image]

sityware, 2023-11-10:
I have found a workaround, also thanks to @mottscanley

Group Outer Walls - Off
Extra Skin Wall Count - 0
Wall Line Count - Add +1

This now only goes over the holes once.

Perds122, 2024-11-26:
Seems this issue has not been solved.
the setting of 'Group Outer Walls' is affecting 'Optimize wall printing order'
If 'GOW' is enabled 'OWPO'  has no affect