Ultimaker/Cura #16167: "Brim distance" not working when using "Per Model Settings"

mannsoft, 2023-07-16:
### Cura Version

5.4.0 and 5.3.0, but not with 5.1.0

### Operating System

Windows 11

### Printer

Creality Ender 3 V2

### Reproduction steps

1. Added an object (.stl) and set "Build Plate Adhesion" to use "Brim".
[image]

2. Added a custom support overlapping a specific area of the first object, and under "Per Model Setting" changed it to "Modify settings for overlaps" and onlly set "Brim Distance" to be 2mm.
[image]

3. Sliced the object and brim is not respecting the 2mm distance on the designated area, instead, the whole object  has a brim
[image]

Also tried the opposite, main object has 2mm Brim Distance and second object has "Per Model Setting" using "Brim Distance" of 0mm. Not working as well
[image]

Also tried using "Infill mesh only" and "Cutting mesh", not working as well.

I also tested using the second object (Per model setting) as another .stl (basically a block) instead of Custom Support. Also didn't work.

And finally, tested using Cura 5.1.0, it worked as expected. 

### Actual results

Final cutting doesn't respect Per Model Setting when using Brim Distance.

### Expected results

Expect the final cut to have two settings of Brim in two different areas: one with Brim Distance of 2mm and another with Brim Distance of 0mm.

### Add your .zip and screenshots here ⬇️

[cura.zip](https://github.com/Ultimaker/Cura/files/12064102/cura.zip)

mannsoft, 2023-07-16:
Just to clarify, the issue happened on Cura versions 5.4.0 and 5.30, but it worked as expected on Cura 5.1.0.

MariMakes, 2023-07-27:
Hey @mannsoft,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I recognize this issue. 🔎
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14862 where the Brim Distance is not working in the per object settings. 

We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14862

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-07-27 by MariMakes