Ultimaker/Cura #16169: Sharp

11sharp, 2023-07-16:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

Instal CURA 5.4.0

### Actual results

CURA 5.4.0
the program starts and stops working during the initialization phase of the plug-in package manager

### Expected results

open the CURA 5.4.0 program

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-07-16:
Thanks for the report.
Without a log file there can't be an investigation.

MariMakes, 2023-07-26:
Hey @11sharp,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

@GregValiant is correct, we don't have enough information to troubleshoot. This might help instead:

I have a couple of questions to narrow it down.
	1. Do you have multiple monitors? Does Cura start if you disable one of the monitors?
	2. Do you have two monitors placed above each other? Can you try this workaround from the community? https://community.ultimaker.com/topic/42633-cura-522-stopped-working/#comment-318105
	3. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
	4. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
	5. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665
	6. Are you the Administrator on your device? Could you try to right-click the shortcut and "Run as administrator"?

Can you let us know if any of these work for you?

If it doesn't work we'll need more information from you like your log files. 
You can find your log file here:
Windows: %APPDATA%\cura\<Cura version>\cura.log

closed 2023-08-10 by github-actions[bot]