Ultimaker/Cura #16171: Cura Quit Unexpectedly on Macbook Pro M2 Pro

skorecki, 2023-07-16:
### Cura Version

5.4.0

### Operating System

MacOS Ventura (13.4.1)

### Printer

Creality Ender 3 v2

### Reproduction steps

Cura application is running on MacOS without any active end user input

### Actual results

Cura 5.4 unexpectedly quits.

### Expected results

The application should remain running without incident

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/12064595/cura.log)

skorecki, 2023-08-04:
This continues to happen regularly.  Frequently, when waking up the laptop from sleep, there is a message about Cure unexpectedly quitting.

MariMakes, 2023-08-08:
Hey @skorecki,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you and to hear your Cura keeps crashing😞

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15161
We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15161

closed 2023-08-08 by MariMakes