Ultimaker/Cura #16174: Crash while attempting to slice.

MicahRummy, 2023-07-16:
### Cura Version

5.3.1

### Operating System

Windows 11, Surface Pro 7

### Printer

Ender 5 Pro

### Name abnormal settings

I changed line width for .6mm nozzle, inital layer height, layer height, initial layer speed, speed, temperature, skirt, etc.

### Describe model location

Center of plate, rotated 180

### Describe your model

Its just a hollow tampered cylinder designed in fusion 360.

### Add your .zip here ⬇️

I included the design files as well in two different formats and as an stl as well as my cura profile.
[Round Trash.zip](https://github.com/Ultimaker/Cura/files/12064695/Round.Trash.zip)

GregValiant, 2023-07-16:
Thanks for the report and the file.
There is a persistent bug that can keep a good model from slicing.  You found one of the workarounds (Initial Layer Height) the other two are rotating the model and moving the model away from the printer midpoint.
I rotated the model 90° and it sliced at .16 Initial Layer Height.
I'll mark this as a duplicate and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic