Ultimaker/Cura #16184: Multi Monitor Support (Nahimic service)

bryanhunwardsen, 2023-07-17:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

N/A

### Reproduction steps

Install Cura on a multi monitor machine (where Nahimic service is already installed and running)
Launch Cura.
Move Cura to any of the non-primary displays (identified/assigned @: Windows=>System=>Display)
Close Cura
Relaunch Cura

### Actual results

Cura begins to render window on the non-primary display it had been previously closed on.
Cura HARD hangs with a white screen, and must be force quit from Task Manager.
Cura is no longer launchable in this state without some variation on generally unacceptable work-arounds.

### Expected results

Cura should be able to launch/run/close then re-launch in a multi-monitor environment with Nahimic Service present and running.

The following are the known recovery options, all of which are extremely problematic and only do a one time mitigation and are not sustainable, none of which is easilly apparent when encountering the issue:

- Disconnect/Disable all but one monitor.
- Stop/Uninstall Nahimic Service (Breaks Audio for the computer)
- Uninstall and Reinstall Cura (Does not work all the time if Windows last close state cache is still referenced to the non-primary monitor)

Additional info can be found in the discussion here: https://github.com/Ultimaker/Cura/discussions/15098#discussioncomment-5491004

I have been made aware that Prusa Slicer has been impacted from the same issue, and that it at least detects the issue and produces a warning (possibly a link to a faq/git bug) of the know issue/mitigations/recovery/work-around). Ity appears they attempted to addresds in a similar manor as I prescribe below: https://github.com/prusa3d/PrusaSlicer/issues/5573

Nahimic Service is bundled Audio software from MANY top desktop motherboard manufacturers, including ASUS, MSI, etc . It does not interfere with anything else on my computer. I am not putting it past that software doing something dirty/careless causing the issue, but Cura should handle this better as this issue has been around for at least 4+ years.

If Cura cant implement a true fix that resolves the issue entirely(but you really should try), at a minimum, detection should be implemented, the user warned/linked to a detailed faq about the issue, recovery steps defined, or better yet allow for the selection of a "clumsy but possibly effective" workaround if Cura detects Nahimic running to clear/reset/request/force Windows to load Cura to/from the (Current) primary defined Monitor (I think Prusa tried something similar back in 2020!!!). 

### Add your .zip and screenshots here ⬇️

No log currently available

MariMakes, 2023-07-27:
Hey @bryanhunwardsen,

Thanks for raising this👍
I'm very aware of the issue with the Nahimic device, I have to supply the workaround quite a number of times. 

I wasn't aware that PrusaSlicer have a potential fix, thank you so much for showing us 🤩 
I'll bring it up to the team to see what they can do with this information. 
I really hope this helps them to resolve the issue. 

Fingers crossed 🤞

bryanhunwardsen, 2023-08-14:
Prusa Slicer (and it's forks) display this message on start-up:
[image]

Here is an anchor to their code that addresses it with detection, user notification, and their Windows Start position reset work-around. https://github.com/prusa3d/PrusaSlicer/blob/287e1e28cd6de02585671d7f74d4e36fddd46674/src/PrusaSlicer.cpp#L720

It would be nice to have some talented engineer block injection of these DLL's  in the first place though.