Ultimaker/Cura #16185: Cura 5.4.0 cannot update settings

wernight, 2023-07-17:
### Cura Version

5.4.0

### Operating System

Linux Crostini (default Debian based)

### Printer

Not relevant

### Reproduction steps

 1. Start Cura (can repro from old or from fresh install without prior settings)
 2. Try to edit custom settings

- Affected version: `UltiMaker-Cura-5.4.0-linux.AppImage` (note that I cannot test the "modern" alternative on my system)
- Working version: `UltiMaker-Cura-5.3.1-linux.AppImage` and prior.
- 
Demo video attached.

### Actual results

Enter, tab or clicking somewhere else should update the setting.

### Expected results

Should update them but does nothing. Also if I upgrade from old settings, dismissing the splash screen is not saved either (i.e. it always shows up again and again).

### Add your .zip and screenshots here ⬇️

[cura-bug-av1.webm](https://github.com/Ultimaker/Cura/assets/145667/064ece23-6cf1-48dd-8c4c-501924d2a3a0)
[cura.log](https://github.com/Ultimaker/Cura/files/12068994/cura.log)

MariMakes, 2023-07-26:
Hey @wernight,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

It sounds like a very irritating issue indeed. It's the first time I've seen this report, and I'm on Windows so I also can't help reproduce it. Your video however is quite insightful. 

I'll bring it up to the team to see what they can do to improve it.

Fingers crossed 🤞

MariMakes, 2023-08-02:
Hey @wernight,

I showed your issue to the team. 
Can you confirm if you used the app image provided by us?
Or did you use one provided by the system package manager?

They believe you might be missing a package that allows you to change the values. 
Does this help you get unstuck?

MariMakes, 2023-08-07:
Sorry to hear you are still stuck 😖 
I have another question that can help with troubleshooting.
Are you using Wayland or X11?

wernight, 2023-08-11:
Crostini runs Sommelier which can run either I think. I'm not sure to be frank which it runs.

MariMakes, 2023-08-28:
Hey @wernight,

I apologize but we are out of suggestions to help you.  To be able to focus on the most requested and needed features and bug fixes from our community and from UltiMaker, we have decided to therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-08-28 by MariMakes