Ultimaker/Cura #16186: Skirt

TomEmmelot, 2023-07-17:
### Cura Version

5.4.0.-beta1

### Operating System

Windowns 11

### Printer

Vertex Delta K8800

### Reproduction steps

Skirt is printed in more layers, why?

### Actual results

Skirt is printed in more layers, why?

### Expected results

Only 1 layer to print.

### Add your .zip and screenshots here ⬇️

Skirt is printed in more layers, why?

mbenkmann, 2023-07-18:
I don't think it's okay to change this default for user-generated profiles. Now I have to manually go into all my profiles and change this setting back.

mbenkmann, 2023-07-19:
I don't have a "definition file". Is that an internal file of Cura that, in my case, would be contained within the big .AppImage blob?

GregValiant, 2023-07-19:
All printer definitions start with the "fdmprinter.def.json" file.  Many printers have their own definition file that "inherits" the "fdmprinter" file and alters settings to make the definition specific to a particular machine.  On my Windows 10 installation there is a **"vertex_delta_k8800.def.json"** definition file.  You _could_ alter the fdmprinter.def.json file but that is not advised when there is an actual printer definition file (like in your case).

On my Windows system the definition files are in "C:\Program Files\UltiMaker Cura 5.4.0\share\cura\resources\definitions".

mbenkmann, 2023-07-19:
That's not an option for me because I use the Linux .AppImage. It's one big blob that contains everything, including the resource files. It does not get extracted to the file system. Sure, there are ways of modifying an .AppImage, but it's obviously not an intended way to change this setting. And when Cura 5.4.1 comes out I'd have to repeat the whole procedure.

GregValiant, 2023-07-19:
You could put in a feature request (which I would support) to change the Skirt Height default to "1".  It will be a couple of months before Cura is updated again.

GregValiant, 2023-07-19:
I can put up with vents and rants so long as they stay on topic.  If a person starts to ramble I loose interest and change the channel.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant