Ultimaker/Cura #16188: Missing tree supports on cura 5.4

Warsenius, 2023-07-17:
### Cura Version

5.4.0

### Operating System

win 10

### Printer

ender3

### Reproduction steps

load project file and slice.

https://i.imgur.com/KIsOEvP.png
https://i.redd.it/6c2e7iqnsicb1.png

On the left i got Cura 5.3 and on the right is 5.4

in the blue circle you can see the pointy lower end of the shield (printing a miniature) being supported. but on the 5.4 version its not getting a support.

file: https://www.thingiverse.com/thing:5921500

printing on a angle for better layer adhesion and so tree support can get everywhere without ever needing to start on the base of the mini.
project file: https://www.mediafire.com/file/h5i9tlfqo70xc29/warforged_with_missing_supports.3mf/file

### Actual results

some parts of the model dont get supported.

### Expected results

supported flying areas

### Add your .zip and screenshots here ⬇️

https://www.mediafire.com/file/h5i9tlfqo70xc29/warforged_with_missing_supports.3mf/file
@ThomasRahm

ThomasRahm, 2023-07-17:
It seems to me that this bug is not in the tree support code.
I am not getting an overhang for the small part, and the tree support only supports areas that were detected as overhangs
If the support overhang angle is reduced (e.g. to 55) it will be detected as overhang.

It seems to me that the issue could be in `computeBasicAndFullOverhang` in `support.cpp` and may be related to `smooth_height` as when i set it to `0.2`, the overhang will be detected.

MariMakes, 2023-07-27:
Hey @Warsenius,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞