Ultimaker/Cura #16190: 70GB of RAM used during slicing. Never got finished.

PeopelCastel, 2023-07-17:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Creality ender 3 pro 

### Reproduction steps

imported:  https://www.printables.com/model/214913-woven-planter/files
Changed walls and som normal settings i do every day like infill. 

Preced slice.
My entire pc slows dows and uses over 32GB of ram and over 70GB with SSD as RAM 

Then i dealeted the original file and replaced it with the other files provided. the same profile 
Changed model for somthing else. (muliple) one file was alot larger and it worked fine. 
When i closed cura the RAM usage drop to about 4.8GB for my entire computer. Only crome for octoprint and cura 

### Actual results

alot of ram usage. like 70GB ish and a super slow pc. ( it is a powerful desktop) 

### Expected results

Sliced file with a reasonable timeframe and a usable gcode. 

### Add your .zip and screenshots here ⬇️

[Error files.zip](https://github.com/Ultimaker/Cura/files/12071991/Error.files.zip)

Asterchades, 2023-07-18:
I'm unable to replicate this with what you've provided. Loading the model increases memory usage by around 110MB (briefly an extra 100MB during the load itself), during slicing Cura's memory usage increases by around 110MB with a further 100MB or so occupied after the slicing has completed, and the whole process takes about 16 seconds on a 3900X. Despite my best efforts in abusing Cura's buffers by reloading the model repeatedly I was unable to get memory usage beyond 3.8GB active (6.3GB commit), for a total system memory usage of 14.7GB out of 32GB, and slicing speed was not noticeably impacted.

To that end you'll likely need to provide a full Project file for further diagnostics - not just a print profile and an STL.

PeopelCastel, 2023-07-18:
I tried again and was able to reproduce it. Even worse than last time. And I just rebooted my pc for the sake of it. I just opend cura, added the file and aliced it with the included profile. Give any instructions if you want any other files/logs

PeopelCastel, 2023-07-18:
here is everything plus a recording of what i did. i used my phone to not afect my pc. 

Computer specs: 
Device name	Marius_Desktop
Processor	11th Gen Intel(R) Core(TM) i5-11400F @ 2.60GHz   2.59 GHz
Installed RAM	32.0 GB (31.8 GB usable)
Edition	Windows 11 Home
Version	22H2
Installed on	‎28/‎09/‎2022
OS build	22621.1992
Experience	Windows Feature Experience Pack 1000.22644.1000.0

System type	64-bit operating system, x64-based processor

Vidio: https://youtube.com/shorts/UdYS6FdanTs?feature=share
Files 
[Updated error files (3).zip](https://github.com/Ultimaker/Cura/files/12081077/Updated.error.files.3.zip)

Asterchades, 2023-07-18:
There it goes! It's only the commit size (ie How much is being asked for) rather than the active working set (ie How much is being used), but that really shouldn't be spiking up so high - I'm seeing the CuraEngine process at just over 61GB for a total commit of 85GB, plus of course the virtual memory thrashing that entails.

First observation is that changing the layer height or rotating the model around the Z axis let it work just fine. This is a common "fix" to some existing faults, but normally these faults will manifest by way of a slicing error - not by massively chewing up system resources. As such this may count as a duplicate, but the unique behaviour makes me think it might not be.

Second observation is that the model is faulty. It's got far too much geometry for me to manually inspect it (SketchUp had a coronary just trying to load it, much less analyse it) but 3D Builder found something and repaired it... eventually. Your project settings were able to slice this repaired model without issue, potentially making it a model issue. But again this is extremely unique behaviour (to my knowledge) so I'm loathed to write it off as such.

I'll flag this one for further inspection from someone more knowledgeable about the inner workings of Cura than I am.