Ultimaker/Cura #16193: Ender 3 Bed Temp jumps over 100

TLison, 2023-07-17:
### Cura Version

5.3.0

### Operating System

Windows 11

### Printer

Ender 3

### Reproduction steps

Print anything thicker than 5mm.

### Actual results

the bed temperature increases to 110 degrees when set to any temperature. Had it set to 60 and it just jumps to 110 after a while.  I have not caught which layer but it happens before it gets to 5mm.  (so before layer 50)

### Expected results

for the Bed temp to stay the same thru the entire print

### Add your .zip and screenshots here ⬇️

[CE3E3V2_PhoneStand.zip](https://github.com/Ultimaker/Cura/files/12072503/CE3E3V2_PhoneStand.zip)

GregValiant, 2023-07-17:
Thanks for the report.
This is absolutely a hardware problem.  It is likely your thermocouple has died stuck in the "On" position.
Your  StartUp Gcode doesn't mention temperatures so Cura adds them right before the StartUp.  This is from a Gcode I made from your project file:
;Generated with Cura_SteamEngine 5.4.0
**M140 S60**
M105
**M190 S60**
M104 S220
M105
M109 S220
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
At the end of the print will be "M140 S0" that turns the bed off.
I'm going to remove the bug label but I'll leave this open so you can provide a gcode file.  If it looks like a Cura problem we can put the label back on, but I'm sure you have a hardware problem.

TLison, 2023-07-19:
That is very odd that Prusa Slicer does not have the same issue on the same printer.  Also I have a new thermocouple on it.

 

Tom Lison

General Manager

Milwaukee PC Inc.

 

(920) 830-9477

***@***.*** ***@***.***> 

MilwaukeePC.com <https://milwaukeepc.com/> 

330 N Westhill Blvd., Appleton, WI 54914

 

Facebook <https://www.facebook.com/MilwaukeePC/>

GregValiant, 2023-07-19:
_"Also I have a new thermocouple on it."_
Yes.  Think about that.

If you open a Gcode file in a text editor and search for M140 and M190 you will find one at the beginning of the file and an M140 S0 at the end of the file.  The normal situation is that there are none between the ;LAYER:0 line and the Ending Gcode section.  A normal situation is for the bed to be turned on in the StartUp Gcode (or just before) and turned off in the Ending Gcode.
There are two things that could add an M140 line amongst the laeyrs:

- You have the post processor "ChangeAtZ" running and you are changing the bed temperature at some layer.
- You have the Initial Layer Bed Temperature set to a low number, and the regular "Bed Temperature" is set over 100.  That would put an M140 line at the end of ;LAYER:0.

TLison, 2023-07-19:
Thanks for your help.

 

Just Odd that this happens on every print using Cura,  however does not occur on any print I slice with Prusa.   The Actual LCD shows the Temp it is set to and the actual temp.  The set to temp jumps to 110 which triggers the bed temp to rise.  I will do more debugging to narrow it down closer to what when I have time.

 

 

Tom Lison

General Manager

Milwaukee PC Inc.

 

(920) 830-9477

 ***@***.***> ***@***.***

 <https://milwaukeepc.com/> MilwaukeePC.com

330 N Westhill Blvd., Appleton, WI 54914

 

 <https://www.facebook.com/MilwaukeePC/> Facebook

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant