Ultimaker/Cura #16194: rough day_checked

Bellhop55, 2023-07-17:
### Cura Version

5.4

### Operating System

Windows 11

### Printer

Anycubic Vyper

### Name abnormal settings

I changed from Generic ASA to Generic PLA

### Describe model location

No

### Describe your model

A woman sitting on 3 stacked tyres

### Add your .zip here ⬇️

[AV_rough day_checked.zip](https://github.com/Ultimaker/Cura/files/12074434/AV_rough.day_checked.zip)

GregValiant, 2023-07-18:
Thanks for the report and the file.
I couldn't slice it with your printer active.  I switched to my Ender and it sliced at 0.08mm Layer Height but not a 0.06.
I didn't find any errors in the model.  Moving and rotating the model didn't help.

I looked through the definition file for you printer but I didn't see anything that would keep this from slicing.
The Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic