Ultimaker/Cura #16199: Slicing taking too long, or never completes

rpersak, 2023-07-18:
### Cura Version

5.4.0

### Operating System

Windows 11, Geforce 3070TI

### Printer

Creality Ender 3 pro, klipper, rpi

### Reproduction steps

Made a model in Fusion 360. It is a model with mesh on the bottom.
Opened it in Cura and started slicing. 

### Actual results

Over an hour later, it is still slicing model.
Prusa slices it in few seconds...

### Expected results

Slicing in few seconds.

### Add your .zip and screenshots here ⬇️

[CE3PRO_Posoda z mrezo v0.zip](https://github.com/Ultimaker/Cura/files/12077396/CE3PRO_Posoda.z.mrezo.v0.zip)

MariMakes, 2023-08-08:
Hey @rpersak,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

This report is really puzzling me 😵 
I'm running into the same long slicing times for Cura 5.4, it seems to be a quicker in Cura 5.0.
[image]

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. Fingers crossed 🤞

rpersak, 2023-08-09:
Ok, thank you for info.

the-moog, 2025-04-15:
Similar problem here. So re-opening this issue rather than starting another. Oddly it sliced before I removed some features, now it fails.  Not sure what to change.
Model attached.
Any way to provide debug data that might help solve [this?]

[console-plinth-v3-wont-slice.zip](https://github.com/user-attachments/files/19756907/console-plinth-v3-wont-slice.zip)