Ultimaker/Cura #16200: 5.4.0 does not start on WIN10

energiebrand, 2023-07-18:
### Cura Version

5.4.0

### Operating System

WIN 10 64Bit / NVIDEA GTX1050Ti / 32GB RAM

### Printer

nothing possible

### Reproduction steps

My current working version is 5.2.1.

When I try tolaunch 5.4.0 the launcher stops at "Updating configuration". After that step nothing anymore happens.
Did remove the 5.4.0 and tried to install again with same result. I can see, that the processor load is at 5%. So there is no activity anymore within the installer.

### Actual results

Nothing happens; program will not start.

### Expected results

Programm will start

### Add your .zip and screenshots here ⬇️

[2023-07-18 09_46_34-Window.zip](https://github.com/Ultimaker/Cura/files/12077617/2023-07-18.09_46_34-Window.zip)

MariMakes, 2023-07-27:
Hey @energiebrand,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I have a couple of questions to narrow it down.
	1. Do you have multiple monitors? Does Cura start if you disable one of the monitors?
	2. Do you have two monitors placed above each other? Can you try this workaround from the community? https://community.ultimaker.com/topic/42633-cura-522-stopped-working/#comment-318105
	3. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
	4. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
	5. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665
	6. Are you the Administrator on your device? Could you try to right-click the shortcut and "Run as administrator"?

energiebrand, 2023-07-30:
Hi Mariska,

 

thanks for your suggestions but unfortunately nothing helps.

 

1.	I have only one monitor BENQ PD2700U via. HDMI
2.	See 1.
3.	No such SW
4.	I have NVIDEA GeForce GTX1050Ti graphics card
5.	No such Audio device
6.	I tried both, normal user and now administrator

 

The behaviour is always the same. Installation works until the step pops up “Updating configuration”. In this phase nothing happens anymore even if I wait 10minutes.

5.2.2 is the last version which runs perfectly.

 

I think I will stay with the last working release and maybe a new release in the future will ran.

Thanks for help anyway!

 

Regards,

Hardy

 

5. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? #12102 (comment) <https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665>

energiebrand, 2023-08-31:
Hi, problem is gone with the latest update of Windows 10.
Now installation works!
Thansk a lot!
Therefore issues can be closed.

closed 2023-08-31 by energiebrand