Ultimaker/Cura #16203: Bad brim slicing error

TheRealOne78, 2023-07-18:
### Cura Version

5.4

### Operating System

Gentoo GNU/Linux with kernel 6.4.1

### Printer

Ender 3 v2

### Describe model location

Please see comment below

### Describe your model

Please see comment below

### Add your .zip here ⬇️

[CE3E3V2_tripod-Body.zip](https://github.com/Ultimaker/Cura/files/12079805/CE3E3V2_tripod-Body.zip)

I've tried to rotate this piece by 90 degrees, but accidentally selected the wrong axys for a second, rotating it by 15 degrees with the wrong axys. After that, I rotated the wrong axys back to it's original position, and finally rotated the right axys by 90 deg.

After slicing it, I noticed the skirt was wrong (see image below), as if the wrongly selected axys was still rotated by 15 deg:
[image]

After that, I've tried re-slicing it, but no progress was made, with a total wait time of about 30 seconds (which is abnormal). After clicking cancel and slice again, the object sliced successfully in about 3 seconds:
[image]

This is the first time I've ever encountered such an issue. As for that, I don't really know if the error is really reproducible, however my little developer knowledge tells me that something is wrong in the build plate generating system on how it updates the axys.

*Note: I have a "Create JPEG thumbnail" post processing script for creating a thumbnail of the model for my printer.*
*Note: This is Modern Linux build with AppImage, x86_64*

GregValiant, 2023-07-19:
Thanks for the report.
I think when you rotated the model and "picked the wrong axis" it ended up with 1 corner up off the build plate by a little bit.
It was enough to let Cura put the skirt under the corner of the model.  In the Cura preferences is the setting "Automatically drop model to build plate".  If that is turned off then what happened is a possibility when rotating a model.

That part appears too be "expected behavior" as Cura was putting the skirt at your "Skirt Distance" away from the portion of the model that was in contact with the build plate.
Here I skewed one part on purpose.
[image]

The other part "The model wouldn't slice" I can't reproduce.  It slices fine for me as it sounds like it did for you (albeit after a couple of tries).
There are quite a few bug reports regarding models that won't slice.  I'll leave the bug label on but I'll close this as the Cura team has lots of examples where "not slicing" is repeatable and this is a duplicate report.  I hope you understand.

TheRealOne78, 2023-07-19:
Hi @GregValiant 
I see that the cura slicer did indeed generate the skirt correctly, as I can see by my previous screenshot of the supposedly bad generated slice on the bottom-left corner of the last layer.

However, I still think this behaviour is abnormal because of one main reason:
I didn't numerically typed how the degrees should change, but used the "circular-arrows" available from the "Rotate (R)" submenu.

When you rotate the object with that feature, cura is supposed to rotate the 3D object exactly 15 by 15 degs, not less, not more.
As I mentioned before, I accidentally rotated the object with the wrong axys by 15 degs, and repaired my mistake afterwards by rotating it back by 15 deg, but with the "circular-arrows", not numerically typed.

So finally, cura shouldn't have leaved that corner in the air, unless the "circular-arrows" (please forgive me if I name these arrows wrongly) did not rotate the object by 15 degs, but more or less.. by meybe +-0.1 deg or so.

I aknowledge the fact that floating numbers are quite a difficult thing to calculate by a binary processor and even harder to implement, but:
1. This only happened (to me) in cura 5.4.0 - which probably means something changed
2. More accurate floating points have already been made and doccumented
3. The "circular-arrows" rotate only by integers (15), not floats

So in contrast, I am now very confident that the rotations were not calculated properly.

GregValiant, 2023-07-19:
I'll go ahead and re-open this then.
The Cura team will take a look at this.

I've been the greeter here for a while now and this is the first report I've seen regarding "incorrect rotation".  On a personal note it has never happened to me either.
There is at least one feature request for a text box that would allow the user to manually enter a specific angle.

TheRealOne78, 2023-07-19:
> I've been the greeter here for a while now and this is the first report I've seen regarding "incorrect rotation". On a personal note it has never happened to me either.
Yes, it's kind of a weird issue, however I suspect it's either because it's a new change on cura 5.4.0 which was released just two weeks ago, or it's an issue people don't bother to report since I just tried to re-slice it and worked as expected, or it's really just a low-level floating point bug which was present since before 5.4, but with very rare occurrences.

This issue really isn't troublesome (at least for me), but I figured I'd report this one here and let the Cura team to decide whether it's an issue worth investigating or not.

GregValiant, 2023-07-19:
I took a quick look at the code in RotateTool.py and didn't notice anything obvious.  The SnapAngle is set to 15 in line 49.

self._snap_angle = math.radians(15)

If you happen to touch the Shift Key while rotating, then all bets are off as the Shift Key will toggle "Snap" on and off.

GregValiant, 2023-07-24:
I had never noticed it either.  I just saw the code in the file.
I go ahead and close this as it seems to be all squared away.

closed 2023-07-24 by GregValiant