Ultimaker/Cura #16204: strong lagging after slicing big models

BramboeckDavid, 2023-07-18:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Anycubic i3 Mega S

### Name abnormal settings

It is lagging anyways

### Describe model location

strong lagging continues

### Describe your model

nothing wrong with the model

### Add your .zip here ⬇️

Small model causes no problems with lagging, big ones do after click on preview.
I have a Nvidia Geforce RTX 3050Ti + Intel Core i5 - 11320H with 16GB RAM. I am very sure that my system is not the problem because slicing very big models at PrusaSlicer causes no lagging issues.
I would be very happy if you can help me!
[AI3MSPRO_Lagging Test.zip](https://github.com/Ultimaker/Cura/files/12080135/AI3MSPRO_Lagging.Test.zip)

GregValiant, 2023-07-19:
Thanks for the report.
I don't understand how you are using "Lagging".  Is it slow to slice?  Slow to change from Prepare to Preview?
Your model sliced quickly (~8 seconds) on my cheap Gateway laptop.  Switching to Preview was near instantaneous.

If you could add some more description of the problem it would be helpful.

BramboeckDavid, 2023-07-19:
Thank you for replying.
It is lagging very strong after slicing when i turn around the model or view different layers.
You can see what I mean at the video i droped below.
[Lagging Problem Video.zip](https://github.com/Ultimaker/Cura/files/12091161/Lagging.Problem.Video.zip)

GregValiant, 2023-07-19:
OK, your video system is having trouble updating the view.
I am using a cheap laptop and I don't have that problem but I've seen it as the more complicated the model, the greater the number of lines being displayed, and the harder the video system has to work.

If you have an onboard video system _AND_ the RTX3050 card then maybe Windows has assigned Cura to only use the onboard video system.  If you have the Nvidia Dashboard app you might be able to configure your computer to insure that Cura has access to the RTX.

I'll leave this open and the bug label in place but see if you can determine if it is a hardware or setting issue on the computer.

BramboeckDavid, 2023-07-20:
Thank you so much, it worked!

closed 2023-07-20 by BramboeckDavid