Ultimaker/Cura #16205: Shared heater T1 temp not set after tool change when extruder temps are set the same

PartDaddy, 2023-07-18:
### Cura Version

5.4.0

### Operating System

Win 10

### Printer

I am the SeeMeCNC factory using a custom profile

### Reproduction steps

1. Open the SeeMeCNC factory project file for Cura 5.4.0 testing. This is in testing and works with a tweak. 

https://github.com/seemecnc/slicing-profiles/tree/master/SE300/Cura/Cura%205-4-0/SMC%20300mm%20Bed/050%20nozzle%20cura540 

2. Notice Extruder 1 is set to 200C, and Extruder 2 is set to 200.1 C
3. The g-code output will work
4. Now set Extruder 2 to the same temperature as extruder 1
5. The g-code no longer outputs temperature settings at tool change and print fails.

### Actual results

no T2 temp setting if both extruder 1 and 2 are set with the same temp.

### Expected results

Tool temperature gcode should be output regardless of same temp settings for extruders. 

### Add your .zip and screenshots here ⬇️

The above description should quickly produce repeatable results. 

[SeeMeCNC-CURA540-Extruder2-Temp-Bug.zip](https://github.com/Ultimaker/Cura/files/12081707/SeeMeCNC-CURA540-Extruder2-Temp-Bug.zip)

GregValiant, 2023-07-18:
Thanks for the report.
If the temperatures are the same, and the heater and nozzle are shared, why would it be required to add an M109 line?  The reason the line appeared in the second gcode file was because the temperatures are different.  That is as it should be.

Line 217 in the Junk file is:
T-1  P0   ;======DESELECT ALL TOOLS
Without a project file I can only surmise that the printer is seeing that line and is de-selecting all tools AND is turning the heater off at that point (do to a firmware instruction).  There are no "M104 S0" lines in the gcode that would tell the printer to turn the heater off.  (Cura didn't turn the heater off so how would it know to turn it back on?)

That line appears to be within your Extruder StartUp Gcode.  Cura does not call out non-existent tools.
What firmware are you running?  RepRap does appear to allow calling out non-existent tools and when it is done, it sets the previous tool to the Standby Temperature as noted in G11.  It would appear that the T-1 command would shut off _any and all_ heaters.  The slicer wouldn't know that.

You appear to have a special situation and you have found one workaround (200.1°).  Another option might be to remove that "T-1" line from the Extruder Startup Gcode.  If the T-1 P0 is required so that no macros run at tool change, then maybe adding an:
M104 S{material_print_temperature}
line to the Extruder StartUp Gcode after the T-1 would help.

I don't see a bug here but rather "expected behavior".  I'll remove the bug label for now.  Let me know if you feel that is premature and I'll leave this open so you can respond.

PartDaddy, 2023-07-25:
Thank you for such a fast response. This can be closed. I find the solution of adding code to the beginning gcode.

Cura 4.6.2 outputs the temp every time, even when temp settings are the same.

closed 2023-07-25 by PartDaddy