Ultimaker/Cura #16209: Enable Infill option missing

Davos2k, 2023-07-18:
### Cura Version

5.4

### Operating System

Windows 11

### Printer

Artillery Sidewinder X2

### Reproduction steps

I have Upgraded from 5.2.2 to 5.4 and now the option to enable infill is missing, I have uninstalled the software and deleted all files for the program and reinstalled but the option is till there. Even if I increase the infill % it always slices without the infill. I have checked through all the options for this setting and nothing allows me to turn this feature on so my prints are all hollow.

### Actual results

option is still missing

### Expected results

I should be able to enable this setting

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-07-18:
Thanks for the report.
There is no option to "Enable Infill" and I don't believe there ever has been.  When the Infill Density is >0 then there will be infill.  If it is =0 then no infill will be generated.

Next to the Setting Search box is a button with three lines on it.  It is the Setting Visibility tool.  Set the visibility to "All".  You can also click on the "equalizer" icon on the right end of the "Infill" toolbar to bring up the visibility settings.  There are 20 infill settings and you are only seeing two of them.

This isn't a bug and I'll remove the label and close this.  If you think that is premature then let me know and we can revisit this.

GregValiant, 2023-07-19:
OK I'll re-open this and the Cura team will take a look.

BTW I have not heard of anyone else with this problem.  That doesn't mean it isn't a problem, but it might have something to do with your system or installation.