Ultimaker/Cura #16210: Empty Lines in .gcode Causes Printer to Stop Print

Erabior, 2023-07-18:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

AON3D M2+

### Reproduction steps

1) Slice part
2) Open part.gcode
3) Observe empty lines with no code or comments
4) Upload to printer
5) Print part.gcode

### Actual results

Printer starts first tool change operation but the print stops and the status indicates the print is completed but also displays X% complete (X< 100%, typically less than 20% depending on size of part).

### Expected results

The print should continue with tool change and continue printing until the part if completed.

### Add your .zip and screenshots here ⬇️

I sent the .gcode of another very small file (testing DSM CF-PA with Infinite Solution AquaSys 120) and they were able to run the file if they removed all empty lines from the .gcode.

This image shows the empty line causing the issue.
[Cura gcode Line Spacing Error.zip](https://github.com/Ultimaker/Cura/files/12083748/Cura.gcode.Line.Spacing.Error.zip)
[image]

These images show the code in "Machine Settings" highlighted by selecting all (ctrl + A) in the text box.
[image]
[image]

GregValiant, 2023-07-18:
Thanks for the report.
Occasionally a post processor might leave an empty line.  For example if you happened to be removing all the M105 lines using Search and Replace without taking out the newline character at the end of the line, then only the new line character would remain.  You would get something like this which I've copied directly from the front end of your gcode: 
;Generated with Cura_SteamEngine 5.4.0
T0
M140 S115

M190 S115
M104 S275
M104 T1 S150

M109 S275

M109 T1 S150
M82 ;absolute extrusion mode

M104 T2 S80
G92 E0
G91 ;relative mode

There are 4 empty lines there that should have M105 in them.  You need to remove the newline character as well.  The same thing is happening after the Extruder StartUp Gcode,...there is M105 missing but the newline character remains.

Search = M105\n
Replace = 
Use Regular Expressions = Checked

Without a project file I have to guess, but this does appear to be an issue with the way you are implementing Search and Replace.  I'll leave the bug label on this for now.  If you will provide a Project File it would be helpful.

BTW:  Within that Extruder Startup code you have G1 E20 F60 as a purge.  Gcode commands are in mm/minute so that's 20mm oF filament at 1mm/second. 
Because the G1 commands that follow don't change that speed, the following "wipe" commands are also at 1mm/sec.  All-in-all, that looks like a really slow tool change.

Erabior, 2023-07-18:
thank you for the reply. I think it's awesome how experienced some people are that you were able to correctly deduce that I was removing M105 using post processing. I also was removing M107 and replacing M141 with M104 T2. I have updated the post processing scripts to: \n for M105 and M107 and M104 T2\n for the M104 T2 script. I did see you mention something about removing the new line character at the end of the extruder start code, but I can't seem to see how I would do that? As far as I can tell there is no new line character at the end of my tool change script. As far as the actual tool change code goes, I don't seem to observe the behavior you would think I would in regard to wipe speed. the purge is slow yes, but the wipe movements are normal in that the actual wiping movements only take a few roughly 5 seconds. there is a noticeable pause after the wiping operation is done but I'm pretty sure that's just the extruder not being exactly at temp and being held up by the M109 command. Now that I think about it, I do not know why that M109 command comes after the tool start code. I feel like it should come before the tool start code. I am unsure, but I have had 2 instances (in the past 2 days) where T1 ate my AquaSys and the print failed. I think this could (might not be) the cause? I am certain it is not a hydration issue as I have probably 800g of silica gel in the dry box and the printer is closed off. Also, I have 8% humidity indicator strips in the dry boxes that are as pink as when I put them in there (blue means >8% humidity present). the worst i have ever seen one of those strips is white after about 1.5/2 weeks when I was taking the ABS out. But I digress. 

TLDR: Thank you, you were right on, Fixed my post processing. However, there is still an empty line after my tool start code. that I am currently using Notepad++ (remove all empty lines) to remove. Finally, the tool change purge is slow, but the movements are normal speed, all 4 moves take like 3-5 seconds total. Pause after wipe I'm assuming is because of the M019 S250, I feel the tool change code should be injected after the M109 to ensure the tool is ready for operation to avoid jamming filament. 

-Thank you so very much

GregValiant, 2023-07-18:
There are settings available when you load the Printer Settings plugin from the MarketPlace.  Two of them are:
Heat Up Speed and Cool Down Speed.
If you do a search for M104 you fill find a line M104 Sxxx Tx up above the tool change.  It's inserted according to those "Heat Up/Cool down" settings.  So the plan is that the nozzle will be up to temperature before the M109 line but if it isn't then the M109 is a safety to insure that printing doesn't start until the hot end is ready.

In regards to the empty line after the tool start code - it should be gone when you fix Search and Replace because it's an M105 line.
G92 E0
T1
G92 E0
**M105**
M109 S215

Since it appears to be sorted out, I'll remove the bug label and close this.

closed 2023-07-18 by GregValiant