Ultimaker/Cura #16211: AnkerMake M5 Slicing Failed

P3dge, 2023-07-18:
### Cura Version

5.4.0

### Operating System

Windows 11 Pro Version	22H2OS build	22621.2048 AMD Ryzen 7 5700G GPU AMD Radeon R9 200

### Printer

AnkerMake 5

### Describe your model

Ironman Fingers, has been succesfully sliced with Cura 5.4.0 and Elegoo Neptune 3 Max

### Add your .zip here ⬇️

[AM5_Fingers.zip](https://github.com/Ultimaker/Cura/files/12088202/AM5_Fingers.zip)

Profile from AnkerMake website
[Cura-5.x-M5-Profile(1).zip](https://github.com/Ultimaker/Cura/files/12088224/Cura-5.x-M5-Profile.1.zip)

GregValiant, 2023-07-21:
Thanks for the report and the files.
I didn't have any trouble slicing your project file and you said you were able to slice it for the Elegoo.
You could try grouping the models together and rotating them on the build plate.  Sometimes change the Initial Layer Height  by .02 or .03 can make a difference.
I'll mark this as a duplicate and the Cura team will take a look.
[image]

P3dge, 2023-07-22:
Hi Greg,
I am pleased to report that i have been able to successufly slice this file by changing the Initial Layer Height from .2mm to .22mm as you have adivised.

Thank you for your support

GregValiant, 2023-07-22:
It's a clumsy workaround and shouldn't be required, but at least it is something to get a model to slice.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic