Ultimaker/Cura #16212: Slicing error

ADYoungberg, 2023-07-18:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Makergear M2

### Reproduction steps

Slicing model for printing

### Actual results

Slicing error message with a button to report the error

### Expected results

Should have completed slicing

### Add your .zip and screenshots here ⬇️

[Toyota_Camry_XV20.zip](https://github.com/Ultimaker/Cura/files/12089473/Toyota_Camry_XV20.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/12089478/cura.log)

GregValiant, 2023-07-19:
Thanks for the report and the file.
The models have some minor errors.  MS 3D Builder was able to quickly repair them.  It still wouldn't slice until I change the "Initial Layer Height" to 0.23.
I'll mark this as a duplicate slicing error and the Cura team will take a look.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic