Ultimaker/Cura #16213: copy+reflect

marqch, 2023-07-18:
### Cura Version

5.4.0

### Operating System

windows 10

### Printer

anet et4

### Reproduction steps

1. open any model in cura
2. copy it
3. use mirror tool and reflect a copy
4. slice

### Actual results

the original model (not a copy) becomes solid (slices with 100% infill)

### Expected results

expected 2 objects with the same settings

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-07-19:
Thanks for the report.
I can't reproduce this.  When I mirror a model it's the same as the original.
[image]

It is possible to force the problem by using the Per Model settings and changing the infill density of one of the models.
Here the left model has 90% density and the right is 20%.
[image]

If you attach a project file it would be helpful.

marqch, 2023-07-19:
Hello. I found what was the problem. It is the "make overhang printable" (experimental) option that gives this result.
[AET4PRO_BoredFrog.zip](https://github.com/Ultimaker/Cura/files/12091957/AET4PRO_BoredFrog.zip)

GregValiant, 2023-07-19:
That one and "Remove all Holes" have their uses, but they are just for specialty situations.

I'll go ahead and remove the bug label and close this then.

closed 2023-07-19 by GregValiant