Ultimaker/Cura #16214: Sovol SV-04 IDEX in Dual Mode Slicing error

JLombardia, 2023-07-19:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Sovol SV04

### Reproduction steps

Take any STL files with supports.
Choose SV-04 Dual Mode printer 
Define to use the supports with the extruder 1 (right) with their own temp.
I'm using extruder 0 (left) at 210º (eSun PLA-SL) and extruder 1 at 190º (eSun PVA)
Slide

### Actual results

The Gcode includes at the beginning the code:
`;Generated with Cura_SteamEngine 5.4.0
T1
M82 ;absolute extrusion mode`

This has a double error. The T1 doesn't finish with ";" and the command selects the wrong extruder for the temp, heating it to the T0 temp. The effect is the Extruder 1 is heated to 220º it has defined value of 190º. The problem is worst when I print with nylon,

### Expected results

That Gcode doesn't show this line as previous versions of Cura. This line is generated internally and I can't delete. The solution is include a T0; line as first line of Initial Gcode section of the machine,
I tested with the old machine profile from Cura 5.3 that works fine and I obtained the same result, This affect only to the Dual Extruder mode.

### Add your .zip and screenshots here ⬇️

[error.zip](https://github.com/Ultimaker/Cura/files/12094407/error.zip)

GregValiant, 2023-07-20:
Thanks for the report.
This is a known bug in the Sovol definition file and the bug report is #16100 .  See Pull Request #16107 for the fix.  You can do the fix on your machine so you don't have to wait for the next version of Cura.

I'll mark this as a duplicate and close it since the bug has been addressed.

JLombardia, 2023-07-20:
The error of #16100 and #1607 are different.
This isn't an temp error of Cura of Sovol. The error is from Cura that write a "T1" line, without ";" at the beginning of Gcode when it must write nothing!!! It only happens with the double extrusion printing.

GregValiant, 2023-07-20:
I can't duplicate it.  A Sovol SV04 Dual Mode set up the way you describe gives this result:

;Generated with Cura_SteamEngine 5.4.0  >INSERTED BY CURA
T1    >>>>>>>>>>>>>>>>>>>>>>>>>INSERTED BY CURA
M82 ;absolute extrusion mode >>>>>>>>INSERTED BY CURA
;SV04 start                                                 within startup gcode
M140 S60;                                                   within startup gcode
M104 S200;                                                 within startup gcode
M280 P0 S160;                                           within startup gcode
G4 P100;                                                      within startup gcode
G28;                                                             within startup gcode
M420 S1;                                                      within startup gcode
M190 S60;                                                      within startup gcode
M109 S200;                                                     within startup gcode
G92 E0;                                                         within startup gcode
G1 X10.1 Y20 Z0.28 F5000.0;                        within startup gcode
G1 X10.1 Y200.0 Z0.28 F1500.0 E15;                 within startup gcode
G1 X10.4 Y200.0 Z0.28 F5000.0;                         within startup gcode
G1 X10.4 Y20 Z0.28 F1500.0 E30;                      within startup gcode
G92 E0 ;Reset Extruder                                     within startup gcode
G1 Z2.0 F3000;                                                  within startup gcode
G92 E0      >>>>>>>>>>>>>>>>>>>>>>>>>>INSERTED BY CURA
Cura never inserts a semi-colon at the end of a line.  That is being done in your StartUp Gcode.  Your "Error.gcode" file is different than you describe.

;Generated with Cura_SteamEngine 5.4.0
T1
M82 ;absolute extrusion mode

;SV04 start
T0;          >>>>>>>>>>>>>>>>>>>startup gcode is changing the Tool Number here.
M140 S60;
M104 S210;

M190 S60;
M109 S210;

M413 S0; disable Marlin power off resume

M280 P0 S160;
G4 P100;
G28; Auto home
M420 S1; auto bed level

;Purge
G92 E0;
G1 X5 Y20 Z0.28 F5000.0;
G1 X5 Y300.0 Z0.28 F1500.0 E30;
G1 X4 Y300.0 Z0.28 F5000.0;
G1 X4 Y20 Z0.28 F1500.0 E60;
G1 E57; retraccion 3mm
G92 E0 ;Reset Extruder
G1 Z2.0 F3000;

G92 E0
G1 F2400 E-0.5
;LAYER_COUNT:249;

Someone has altered your startup gcode.  You can see that the stock SV04 Startup does not have any blank lines, there is no M413 command, and in your ?Error file" the word "retraction" is misspelled.

I will re-open this, but I'll leave the bug label off for now.

- Load a model and set Cura up the way you had it.
- Use the "File | Save Project" command to create a 3mf project file.
- Right click on the file in Explorer and select "Send To | Compressed (Zipped) Folder".
- Post the zip folder here.

GregValiant, 2023-07-20:
This is all about the StartUp Gcode that is within the definition files.  UltiMaker Cura has nothing to do with what is contained within the definition file.  Those files come from either the printer manufacturer, or from a member of the community.
From the SV04 Dual definition file:
    "name": "Sovol SV04 Dual Mode",
    "inherits": "sovol_base",
    "metadata":
    {
        "visible": true,
        **"author": "Sovol",**
Whatever is going on with the StartUp is either direct from Sovol, or a customization by a user.  This cannot be a Cura bug because the UltiMaker company doesn't have any way to check on the accuracy of the definition files of over 300 different printers.

In 5.4 this is what I have for the StartUp G-Code after installing an SV04 Dual printer:

;SV04 start
M140 S{material_bed_temperature};
M104 S{material_print_temperature};
M280 P0 S160;
G4 P100;
G28;
M420 S1;
M190 S{material_bed_temperature};
M109 S{material_print_temperature};
G92 E0;
G1 X10.1 Y20 Z0.28 F5000.0;
G1 X10.1 Y200.0 Z0.28 F1500.0 E15;
G1 X10.4 Y200.0 Z0.28 F5000.0;
G1 X10.4 Y20 Z0.28 F1500.0 E30;
G92 E0 ;Reset Extruder
G1 Z2.0 F3000;

In my opinion, that StartUp Gcode is incorrect.  Cura should be allowed to dictate when the heaters are called and when the tools are selected.  I think that StartUp should be:
;SV04 start
M280 P0 S160;
G4 P100;
G28;
M420 S1;
G92 E0;
G1 X10.1 Y20 Z0.28 F5000.0;
G1 X10.1 Y200.0 Z0.28 F1500.0 E15;
G1 X10.4 Y200.0 Z0.28 F5000.0;
G1 X10.4 Y20 Z0.28 F1500.0 E30;
G92 E0 ;Reset Extruder
G1 Z2.0 F3000;

Removing the temperature lines forces Cura to add the temperatures prior to the StartUp gcode.  The result is like this:
T1
M140 S60
M105
M190 S60
M104 T0 S175
M104 S215
M105
M109 T0 S175
M105
M109 S215
M82 ;absolute extrusion mode
;SV04 start
M280 P0 S160;
G4 P100;
G28;
M420 S1;
etc, etc.
The downside is that the bed leveling is taking place with hot nozzles and that means there will be some oozing.

JLombardia, 2023-07-21:
I don't agree.
The Gcode in cura 5.3 started with:

JLombardia, 2023-07-21:
Sorry, I don't agree.
If I leave the Start Gcode personalization empty in Cura 5.3 it starts with:
    ;Generated with Cura_SteamEngine 5.3
    T0
    ....
If I leave the Start Gcode personalization empty in Cura 5.4 it starts with:  
    ;Generated with Cura_SteamEngine 5.4
    T1
    ....
In both cases using the same profile machine, not the profile included in 5.4. I could understand that Sovol sent to you a defective machine profile, for that reason I'm using the same.

I attach you the profile to test with them. This profile must appear into the Sovol machines and it's called SV04L-Dual_mode (with an L after SV04). The profile uses an extruders with the same name modification.
Using it Cura 5.3 starts with T0 and Cura 5.4 with T1.

Preheat the E1 sometimes it's  a lost of energy because you could use this extruder some hours later for a top layers.
The machine needs to preheat before autohome because it has a cleaning system of the nozzle. If there are cold filament outside the nozzle from last work, the machine forced the movement of the extruder and the motor jump over the belt and lost the calculated position. Autohome only use E0, E1 doesn't have height sensors.

[sv04L.zip](https://github.com/Ultimaker/Cura/files/12117585/sv04L.zip)

JLombardia, 2023-07-21:
About your comment that removing the temperature lines forces Cura to add the temperatures prior to the StartUp gcode. The result is like this:
    T1
    M140 S60
    M105
    M190 S60
    M104 T0 S175
    M104 S215
    M105
    M109 T0 S175 
    ....,
 Yes, it writes but there aren't the printing temperatures. They are the waiting temperatures. When an extruder isn't working it reduces the temp to this value to reduce the possibility of a clog and don't overburn the filament. So it isn't a valid solution.
 

About the matter of this post I have taken the Sovol website profiles for Cura 5.3 and I modified the name so you can import in Cura 5.3 and Cura 5.4 without confusion with the included in Cura 5.4. I attach you.
 With this machine profile you can slice anything with Cura 5.3 and the gcode will start:
 `;Generated with Cura_SteamEngine 5.3
T0`

Same file and same machine profile in Cura 5.4 will start:
 `;Generated with Cura_SteamEngine 5.4
T1`

Same machine profile, same profile, same STL, different starting code. I can't understand that it's a Sovol fault.

[sv04L.zip](https://github.com/Ultimaker/Cura/files/12122726/sv04L.zip)

GregValiant, 2023-07-21:
I added those definition files to 5.4.0 and 5.3.1
I went to my configuration folder and removed all references to Sovol printers from both the extruders folder and from the definition changes folder.  That was to insure a fresh install.
I installed a new Sovol Dual in both versions of Cura.

- In 5.4.0
- "T1" is inserted prior to the StartUp Gcode.
- T1 starts the skirt.
- T1 prints the support
- Switch to T0
- T0 finishes the skirt
- Switch to T1
- Print support with T1

In 5.3.1
- "T0" is inserted prior to the StartUp Gcode.
- T0 starts the skirt
- Switch to T1
- T1 finishes the skirt
- T1 prints the support

So it appears that 5.4.0 starting with T1 adds a tool change that isn't present (in my example) when starting with T0 as 5.3.1 does.  Functionally, that is the only difference I see.  The correct extruders are printing the proper portions of the model.  The preview in both cases is correct.
Using a Brim causes both versions to start with T1.

Is that what it looks like to you?

This is my test model.
[image]

JLombardia, 2023-07-21:
T1 it's the auxiliar extruder and usually is used for supports.
T0 is the main extruder that must print the brint and the piece.

In Cura 5.3 is right.
In Cura 5.4 is wrong the starting. Heat T1 with the temp of T0, and start printing with T1 when it must do with T0, but later Cura corrects it and use T0 for the main and T1 for the supports. Including a simply T0 line in the custom Gcode it solves the problem because "repair" the T1 line.

It's more clear the error if you use a profile with a border and not a brim and the pieze is directly over the bed. The border must be printed with the T0 like the first layer of the piece. In image seems to be fine but the Gcode is wrong by the T1 line of the beginning. And if you decide use the T0 for the brim, the Gcode is wrong too.

[image]

WIth this piece the Gcode;
Activate T1
Overheat T1 with the temp of T0.
Purge with the T1.
Print the border and the first layer of piece with T1
Then heat T1 to the right T1 temp
Print the supports first layer with T1
Change to T0
Print right the second layer of T0.
And all the following is right but the beginning is wrong.

If the initial T1 it would a T0 like 5.3:
Activate T0
Heat T0 correctly
Purge with the T0.
Print the border and the first layer of piece with T0
Then heat T1 to the right T1 temp
Print the supports first layer with T1
Change to T0
And all is right

GregValiant, 2023-07-21:
Alright, I think I have a handle on this.
When {material_print_temperature_layer_0} or {material_print_temperature} is called out in the StartUp Gcode Cura seems to always replace it with the temperature value for **Extruder[0]** rather than the temperature value for **Extruder[current_extruder]**.
When there is no temperature callout using replacement patterns in the StartUp Gcode then the proper temperature is used.
I'll put the bug label back on this and the Cura Team will take a look.
This report may get moved to the "Cura Engine" Github which has it's own bug reports.

JLombardia, 2023-07-21:
Thank you.
Meanwhile I left the T0 in the first line of the custom code.

GregValiant, 2023-07-21:
I added a bug report to the Cura Engine [#1912](https://github.com/Ultimaker/CuraEngine/issues/1912)