Ultimaker/Cura #16215: Elegoo Neptune 3 Pro: Printer Profile overrides material settings

jofland, 2023-07-19:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Elegoo Neptune 3 Pro

### Reproduction steps

1. Install Printer with Profile "Elegoo Neptune 3 Pro" ==> uses profile in file "C:\Program Files\UltiMaker Cura 5.4.0\share\cura\resources\definitions\elegoo_neptune_3pro.def.json"
2. Chose predefined material or define one for your needs (eg. Default Printing Temperature 227, Retraction Speed 27, ...)
3. Watch changes of Printing Temperature and Retraction Speed in Profile Settings window

Cross Check:
Do step 1. to 3. with Printer Profile Creatly Ender 3 S1. 

### Actual results

No changes in Printing Temperature or Retraction Speed. Chosing different materials do not change Print Settings.

### Expected results

Step 3. shows defined parameters (eg. Default Printing Temperature 227, Retraction Speed 27, ...) that differ from default values

### Add your .zip and screenshots here ⬇️

I scanned the file "C:\Program Files\UltiMaker Cura 5.4.0\share\cura\resources\definitions\elegoo_neptune_3pro.def.json" for "temp" and "retract". Then I removed the following lines:

        "material_bed_temperature": { "value": "default_material_bed_temperature + 10" },
        "retraction_amount": { "value": 1 },
        "retraction_retract_speed": { "resolve": 25 },
        "retraction_speed": { "resolve": 25 },

Saved the file with different profile name and imported the newly created printer.

Now steps 1. to 3. show expected behaviour. Material parameters are used in print settings.

==> Could someone please review the printer definition file completely and provide a clean version so that all material parameters are used when changing the material?

Maybe the same issues exist for other neptune series printers.

GregValiant, 2023-07-20:
Thanks for the report.
This isn't an UltiMaker Cura problem as the definition files for the Elegoo printers were authored by "Naruto".  The pull request to add the printers is #14231 .
The UltiMaker Company doesn't have Elegoo printers to check the definitions against.  You would need to contact "Naruto" maybe through Elegoo?  Maybe another community member can help.
I'll leave this open for you but I'll remove the bug label.  I hope you understand.

jofland, 2023-07-21:
Ok. Thanks. I understand.

But I see github as a community platform to solve problems, not only addressing ultimaker as a company.

Maybe the community can help.

jofland, 2023-07-21:
@NARUTOfzr could you please have a look at this issue? What do you think to solve this?

Meanwhile I figured out that the noozle profiles have fixed temperatures too. Maybe something else?

GregValiant, 2023-07-21:
Yes it is and speaking for myself, I help when I can.  In this case I can't as I don't have an Elegoo printer.  Other users will will see this and hopefully will provide a solution.

jofland, 2023-07-21:
I'm an absolute beginner in cura configuration. So far as I understand the config files derive from each other and are able to override parameters the deeper the derivative is. Correct?

Is there a diagram of this dependencies including printer, material, noozle and so on?

Then it would be smart to set settings fix as deep as possible in the derivative tree.

And as far as there are different elegoo neptune 3 models with only different build volumes to derive the parameters from one base model.

GregValiant, 2023-07-21:
Frankly I gave up on trying to understand which overrides what.
The settings have icons next to them.  Those icons can "link" the setting to some other setting.
In the example of print temperature, you can set the temperature to whatever you like.  If it is different than the print temperature in the material file, your override "unlinks" the material profile from your setting profile.  You will see the "fx" icon and the "circle-arrow" icon appear indicating that you have set a custom temperature.  Clicking on the fx icon will re-set the temperature back to the default for the material.  Clicking the other icon will reset the temperature and re-link the setting.

MariMakes, 2023-08-02:
Hey @jofland,

Welcome to the UltiMaker Cura Github 🚀 

I'm just swooping in to mention that we have documentation on where settings come from. 
You can find it here https://github.com/Ultimaker/Cura/wiki/Getting-A-Setting-Value

Printer definition files are one of the highest tiers in the container stack. If you change them by hand, or change them with a profile it's a lot lower in the stack and there is a higher chance that it will not be overwritten by another setting. 

If you want to learn even more about it one of our Developers made a plug-in to analyze it a while a go. 
https://github.com/sedwards2009/cura-god-mode-plugin

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

jofland, 2023-08-03:
I think it's a bug in the standard cura configuration files for this printer. So please keep it open. Every inexperienced user with this printer will run into the same issue.

Maybe @NARUTOfzr wants to change this files soon?

Meanwhile I changed the values in the files for my installation. But I'm not sur if everything is correct.

alexose, 2023-11-27:
Chiming in to say that I ran into this issue the hard way.  I wasted a lot of time trying to debug it...