Ultimaker/Cura #16216: Support Brim is simply a single line with Tree supports

sanslos, 2023-07-19:
### Cura Version

5.4.0

### Operating System

Windows 10 Pro

### Printer

RadZ (Modified M3D Promega)

### Reproduction steps

Generate Support
Support Structure: Tree
Enable Support Brim
Set "Support Brim Width" to some value (e.g. 1.2)

### Actual results

There is s single width line at whatever the the Support Brim Width is from the tree support, but there is no actual brim.  There is the tree, and the single-width "support brim" line, nothing else.

### Expected results

There should be a brim, a continuous layer of material, from the tree support to whatever the "Support Brim Width" distance is inside the tree.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-07-20:
Thanks for the report.
A project file is necessary for an investigation.
There has been a bug reported that occurs when Bed Adhesion is set to "None" and "Support Brim" is turned on.  If you set the Bed Adhesion to a skirt the Support Brim should show up.
Let us know if that works for you.

sanslos, 2023-07-21:
Howdy.
I do have a project file (attached), I didn't include it initially because I suspected I misunderstood what the options do.  From what you've said it seems you have everything you need, please let me know if there is anything else I can do.

So, yes, I have "Build Plate Adhesion Type" set to "None". When I set it to "Skirt" I get the Support Brims.

Thank you very much, that addresses my current needs :-)
[CFFFP_Baby_Hand.zip](https://github.com/Ultimaker/Cura/files/12114980/CFFFP_Baby_Hand.zip)

sanslos, 2023-07-21:
Thank you for your help.  My immediate problem is addressed, I'll close this issue.

closed 2023-07-21 by sanslos