Ultimaker/Cura #16218: Cura failure on complex model.

MTCoalhopper, 2023-07-19:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

MakerBot Replicator

### Name abnormal settings

default settings for printer

### Describe your model

Have successfully sliced for Voxelab Aquila.

### Add your .zip here ⬇️

[Weber DCOE Body Stepped barrels.zip](https://github.com/Ultimaker/Cura/files/12100892/Weber.DCOE.Body.Stepped.barrels.zip)

GregValiant, 2023-07-20:
Thanks for the report and the file.
The model has a minor error that Cura colors in a polka-dot pattern.  It is easy to repair with MS 3D Builder.
[image]

I didn't have any trouble slicing it with the error so I can't duplicate the problem.
[image]

I'll mark this as a duplicate and the Cura team will take a look.

Now to the crux of the matter.
I'm upset that it isn't [my DCOE](https://www.thingiverse.com/thing:4444855).
[image]

After I modeled that up I decided to go with 40IDF's as the downdraft manifold was a lot easier to fab.

MTCoalhopper, 2023-07-20:
I appreciate the quick response.  The DCOE model was chosen because it's
the most complicated part that I've got that can be printed in full scale.
The library where I am setting up a maker space acquired a MakerBot
Replicator Z18, today, and tomorrow we're going to test it with something
challenging. The DCOE was supposed to fit that bill, but I may just used my
own IDF files.

GregValiant, 2023-07-20:
Any chance of you sharing the IDF files?
This is with the downdrafts.
[image]

I originally designed these manifolds for the DCOE and then adapted them to the stock Keihens.  Doesn't run as well as with the Webers.  These are fiberglass with aluminum attachment plates.
[image]