Ultimaker/Cura #16219: Slicing fails in this position

jbcpollak, 2023-07-20:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Creality CR-10S Pro

### Describe your model

got a warning that it wasn't water tight, but previously had been slicing it ok despite the warning.

### Add your .zip here ⬇️

After this I picked the model up repositioned it and had Cura snap it flat to the bed in a new position. The first time I tried this it failed, I tried a second time and it worked.

[CCR10SPRO_console.zip](https://github.com/Ultimaker/Cura/files/12101545/CCR10SPRO_console.zip)

GregValiant, 2023-07-21:
Thanks for the report.
This bug has carried over from 5.3.1 and the Cura team is aware of it.  Moving, rotating the model or a small change to the Initial Layer Height can often fool the bug.
I'll mark this as a duplicate and go ahead and close this.  I hope you understand.

closed 2023-07-21 by GregValiant

jbcpollak, 2023-07-22:
Fine with me, glad you are tracking it.