Ultimaker/Cura #16221: graphic

st0p0r, 2023-07-20:
### Cura Version

5.4

### Operating System

Windows 11

### Printer

BIQU B1

### Reproduction steps

in some places the numbers intersect with the text. this happens on all 5 versions
[image]

### Actual results

hard to read result

### Expected results

text should be clearly visible

### Add your .zip and screenshots here ⬇️

it would not hurt to add the ability to change the font and its size

MariMakes, 2023-07-27:
Hey @st0p0r,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Your issue reminds me of https://github.com/Ultimaker/Cura/issues/15361

Do you happen to have your region set to a country where the language is aligned to the right like Hebrew? I don't see it because the language in my region is set to the left:
[image]

One of our developers indicated that it's not too hard to fix, I'll ask them to take another look. 🤞

st0p0r, 2023-07-28:
Hi Mariska,
you are absolutely right - my region is Israel and I do not intend to
change it to another for the sake of one program.
I will be glad if you have another solution.
regards,
Mark

MariMakes, 2023-08-07:
Quick update from our side 👋

One of our developers was able to resolve it, so it should be fixed in the next release 😊 
You can read more about the fix here https://github.com/Ultimaker/Cura/pull/16270

Thanks again for your report 👍

closed 2023-08-07 by MariMakes