Ultimaker/Cura #16222: New Material - All Profiles Get Lost.

MicahRummy, 2023-07-20:
### Cura Version

5.3.1

### Operating System

Windows 11, Surface Pro 7

### Printer

Ender 5 Pro

### Reproduction steps

I added a new material, and after that all my profiles were gone including the defaults. The selected profile was "Not Supported" literally the name. I went to my old material that I created back in Cura 4, and all my profiles reappear. I tried exporting it and importing it into the new material, but that just duplicates it into the original material I was using.

Problem is that new materials cannot be created whist keeping existing profiles.

### Actual results

See Reproduction steps. I couldnt simply separate everything. Its self explanatory.

### Expected results

I expected all my profiles to be universally shared between all materials like in older versions of Cura.

### Add your .zip and screenshots here ⬇️

[Screenshots.zip](https://github.com/Ultimaker/Cura/files/12102300/Screenshots.zip)

ugurcsen, 2023-07-20:
I tried to update Cura from 5.3.1 to 5.4.0 and I lost all default profiles for Elegoo Neptune 3 Plus—device: Macbook Pro M1 (13.3.1 (a)).
After that, I tried a clean install and reset configs but it did not work but clean install on Windows 11 works well.
My problem looks like related to this bug. Should I open a new issue?

MariMakes, 2023-07-27:
Hey @MicahRummy,

I believe what you are encountering is this issue: https://github.com/Ultimaker/Cura/issues/8229
It's something that a significant amount of people encounter.

If you make your own material you have to make sure that the Material Type matches one of the Generic Materials Exactly else it loses all the quality files. 
[image]

Since you are working with ASA I suggest to set ABS as Material Type. You can change the other fields like Brand and Display Name as much as you want, but the Material Type is needed by Cura to find you the correct quality profiles. You should not see the ABS pop-up in other places. 

We are planning to improve this by asking people to select a material from a dropdown. Would this resolve your issue? 
[image]

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/8229

@ugurcsen, I believe what you are encountering is this: https://github.com/Ultimaker/Cura/issues/16007

closed 2023-07-27 by MariMakes