Ultimaker/Cura #16223: Cura adds weird temp settings before my start-code.

TheNecronomica, 2023-07-20:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 S1 Pro

### Reproduction steps

1. Add model to Cura
2. Change "printing temperature initial layer" to 10°c higher than "Printing temperature".
3. Add "M140 S{first_layer_bed_temperature}" and "M104 S{first_layer_print_temperature}" to start G-code in printer settings before G28; Home.
4. Add "M190 S{first_layer_bed_temperature}" and "M109 S{first_layer_print_temperature}" to Start G-code after G28; Home.
5. Slice.

### Actual results

[image]

Cura adds this part before the start G-code. 

### Expected results

I expect the marked G-code not to be there.

### Add your .zip and screenshots here ⬇️

[cura.log.zip](https://github.com/Ultimaker/Cura/files/12104051/cura.log.zip)

fieldOfView, 2023-07-20:
Instead of `{first_layer_bed_temperature}`, use `{material_bed_temperature_layer_0}`
Instead of `{first_layer_print_temperature}`, user `{material_print_temperature_layer_0}`

For a neat list of all replacement patterns for Cura settings, see this page:
http://files.fieldofview.com/cura/Replacement_Patterns.html

TheNecronomica, 2023-07-20:
> Instead of `{first_layer_bed_temperature}`, use `{material_bed_temperature_layer_0}` Instead of `{first_layer_print_temperature}`, user `{material_print_temperature_layer_0}`
>
> For a neat list of all [...]

This worked!

closed 2023-07-20 by fieldOfView