Ultimaker/Cura #16225: Allow users to add their own nozzle sizes to a machine profile

chipmunk1977, 2023-07-20:
### Is your feature request related to a problem?

On the Cura Users group on Facebook there is a frequent request for help when a user can't  change the nozzle size on a Community- or manufacturer-provided machine profile.  (Several such issues have been reported here too.)
A workaround exists, such as shown in this thread:  https://github.com/Ultimaker/Cura/issues/15190
However, this is quite an ask for some users.

### Describe the solution you'd like

Would it be possible to include a field in the machine profiles which allow a user to change the nozzle size when an existing profile without this option exists?
I would seem to me that changing a nozzle is not particularly uncommon and therefore should be easier for users to integrate into the Cura user experience.  Maybe having the current dropdown for the author's nozzle sources could automatically have an entry, "Use another nozzle size" which would then open a field where this could be entered in by hand.
I have no idea about the coding practicalities of this, but given that Cura handles the change in flow requirements for different nozzle sizes automatically, I don't see a negative side for the user.

### Describe alternatives you've considered

Alternatively a template could be created by Cura for contributed profiles which automatically includes all the standard nozzle sizes.  The contributor includes specific profiles of their choosing, but the other sizes can still be accessed by the user.  (This might actually be easier for the Cura devs to implement.)

### Affected users and/or printers

Any community- or manufacturer-provided profile which doesn't include options for a full set of nozzles.

chipmunk1977, 2023-07-20:
The problem comes about when a contributor has included variants - for example, a 0.4 and a 0.3.  At that point the user can't easily include a 0.6 nozzle.  The user is stuck with the options that the contributor has given.  That's what I am suggesting needs a more simple workaround than the way shown in the Issue i linked to above.

GregValiant, 2023-07-20:
Yea, they are stuck having to alter an existing nozzle file and doing a save-as to create the new size nozzle.

Within the Printer Settings though is the "real" nozzle size box.  The ones in the Machine Settings and in the Nozzle Size dropdown list simply update the Printer Settings Nozzle Diameter and that is the real setting that Cura uses for Nozzle Size.

This is another of those things that documentation is really hard to find.  It's doubtful anyone would actually read a users manual, but if there was a comprehensive one then users could be pointed to it.  Of course then someone would have to keep it updated as well.

FriedSemiconductors, 2023-07-20:
@chipmunk1977 Agree, i like Cura, but this is a much needed feature.
Users should be able to add multiple nozzle sizes in their config, and be able to select/enable the preferred one, before slicing.

For example, the Ultimaker 2 series with Olsson Block only let's you choose from the following sizes:
`0.25mm`, `0.4mm`, `0.6mm` and `0.8mm`

There are many compatible nozzles of different sizes, that are great, but we have to resort to workarounds, because the software doesn't accommodate for this.
One might even call it a design flaw or oversight, this does impact software ergonomics/usability to some degree.
You can often see questions on how to select a certain nozzle size in forums or facebook groups, which would otherwise be unnecessary.

[image]