Ultimaker/Cura #16227: Slicing error on frog model

enewbold, 2023-07-20:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Flsun V400

### Reproduction steps

Loaded the BoredFrog_belly.stl.
Sliced it (with tree supports).

### Actual results

Slicing failed.
[image]

### Expected results

It should have sliced.

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/12112856/cura.log)
[cura.log.txt](https://github.com/Ultimaker/Cura/files/12112861/cura.log.txt)
[FV_BoredFrog_belly.zip](https://github.com/Ultimaker/Cura/files/12112876/FV_BoredFrog_belly.zip)

GregValiant, 2023-07-20:
Thanks for the report and the files.
It wouldn't slice for me either.  I changed the Initial Layer Height to .27 and moved the model away from the center of the bed and it sliced.
I'll mark this as a duplicate and the Cura team will take a look.
[image]

enewbold, 2023-07-21:
I actually got Orca 1.6.3 to slice it with no problems, but the Orca slicer
output is not as good as Cura.

Thanks much,
Ed Newbold

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic