Ultimaker/Cura #16230: Slicing error

williams1964, 2023-07-20:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Anycubic Kobra Pluus

### Name abnormal settings

Switch to fine

### Describe model location

Center of plate

### Describe your model

Yes

### Add your .zip here ⬇️

[Pawns.zip](https://github.com/Ultimaker/Cura/files/12113769/Pawns.zip)
I did change the nozzle to a .2 vs a .4

GregValiant, 2023-07-20:
Thanks for the report and the files.
They wouldn't slice as received.  I checked the models and both show errors.  After repairing the models I still had to rotate them 90° to get them to slice.
I'll mark this as a duplicate and the Cura team will take a look.
This is with a 0.20 nozzle.
[image]

williams1964, 2023-07-21:
I have sliced them before with the nozzle setting at .4 and it was fine. As soon as I change it to .2 I got the error. I have downloaded it again and still get the error. Sent from my iPhone