Ultimaker/Cura #16233: Slicing failed on custom design of hydroponic tower

tommyhp2, 2023-07-21:
### Cura Version

5.3.1

### Operating System

Windows 10

### Printer

Creality Ender 5 Plus

### Name abnormal settings

Just minor changes to temperatur, print speed and wall thickness (3 layers)

### Describe model location

no

### Describe your model

Simpler design

### Add your .zip here ⬇️

Attached file has 2 projects:

CE5P_hydro_t3_spacer **sliced OK**

CE5P_hydro_t3 **failed to slice.**

[CE5P_hydro_t3_bug_report.zip](https://github.com/Ultimaker/Cura/files/12117278/CE5P_hydro_t3_bug_report.zip)

GregValiant, 2023-07-21:
Thanks for the report and the files
I didn't have any trouble slicing the project.  I noticed the model was not watertight so I repaired it with MS 3D Builder.  Then I had trouble slicing it.  Rotating the model allowed it to slice again.
[image]

The same sort of thing happened with the tube detail.  This is the un-repaired model.
[image]

The Cura team will take a look.

tommyhp2, 2023-07-24:
@GregValiant,  Thanks for the feedback.  If you're referring to the slightly short gap (2 mm ?) of the inner walls, that was intentional.  I've tried rotating the model at plus/minus 45 and 90 but still couldn't slice for some reason.  I wished the program did some kind of dump of the error which I think would help the devs quickly identify the error if the it's a bug with the program nor the environment/settings of the app.

tommyhp2, 2023-09-12:
Just tried in version 5.4.0 and it works as expected.  BTW, the slicing is faster than than 5.3.x.  Thanks!  Very much appreciated :)

closed 2023-09-12 by tommyhp2