Ultimaker/Cura #16234: cant add weefun tina2s

tuckem, 2023-07-21:
### Cura Version

5.4.0

### Operating System

windows 10

### Printer

weefun tina2s

### Reproduction steps

open up cura 
tried to add new printer click weefun tina2s 

### Actual results

 cura instantly crashes 

### Expected results

add new printer

### Add your .zip and screenshots here ⬇️

[cura.log.zip](https://github.com/Ultimaker/Cura/files/12117753/cura.log.zip)

GregValiant, 2023-07-21:
Thanks for the report.
I didn't have any trouble loading the printer.
I see in the log file that Cura was not able to find the extruder file "weedo_base_extruder_0.def.json".  It does exist on my installation in "C:\Program Files\UltiMaker Cura 5.4.0\share\cura\resources\extruders" folder.
The Cura team will take a look.

tuckem, 2023-07-22:
oh thanks do you recommend uninstalling then re installing or should I just
add that file I tried pulling all those files from The Cure the refund
comes with which is like 4.1.0 or something like that but when I load them
up in the new Kira it says there's errors I'm just trying to get one slicer
Kira to have all my profiles for printers which I just have the anycubic
kobra Neo , anycubic kobra 2 and  weefun tina2s I have amazing files for
the Cobra 2 and also Neo for Kira they're not my work somebody else made
him uploaded him on GitHub and thingiverse but they're a great profiles if
you like I could share them with you see you guys can add those two
printers also it'd be nice

Thomas Healey

GregValiant, 2023-07-22:
Unzip the folder below.  The "weedo_base_extruder_0.def.json" file goes in "C:\Program Files\UltiMaker Cura 5.4.0\share\cura\resources\ **extruders** ".
The "weefun_tina2s.def.json" file goes in "C:\Program Files\UltiMaker Cura 5.4.0\share\cura\resources\ **definitions**"

[weedo_base_extruder_0.def.zip](https://github.com/Ultimaker/Cura/files/12136238/weedo_base_extruder_0.def.zip)

I don't know that your problem will be fixed, but it might be.

MariMakes, 2023-07-27:
Hey @tuckem,

Welcome to the UltiMaker Cura Github 🚀 

I am able to add a Weefun Tina2S to my Cura set-up, without a crash. 
I agree with @GregValiant, you seem to be missing an extruder definition:
`Exception: Unable to find extruder definition with the id [weedo_base_extruder_0]`
 
Can you let us know if copying his files in your folder end up resolving your issue?
Thank youuu 🙏

tuckem, 2023-07-30:
no still crashes  i followed gregs advice adding both files in
appropriate places

GregValiant, 2023-07-30:
You will probably need to do a full un-install and then re-install.
**NOTE:**  that following the instructions below will wipe out any customizing of materials, or setting profiles, or any printers you may have installed in Cura.  It's "Back to Square One" time.

Before un-installing Cura use the "Help | Show Configuration Folder" command to locate the Cura User Settings folder.  Remember where it is.
Un-install Cura.
Go back to the Configuration Folder and delete the folder "5.4".
Go to the Program Files folder and make sure the "UltiMaker Cura 5.4.0" folder is gone.
Re-install Cura.

MariMakes, 2023-08-07:
Hey @tuckem,

Can you let us know if you've been able to add the printer after uninstalling and reinstalling Cura?

This is how you perform a clean install
	1. Quit the application
	2. In a file explorer window, navigate to:
			My Computer/users/[user_name]/AppData/Local/cura
			My computer/Users//AppData/Roaming/cura
	1. Delete the entire 'cura' folder of your current version. Note: Cura folders of previous versions will be imported if not deleted!
	2. Open Ultimaker Cura. If it has been reset correctly, you will be greeted with the onboarding flow
If you do this you will use any custom profiles. If you want to keep these make sure to make a backup.
You can read more about them here: https://support.makerbot.com/s/article/1667412704080

closed 2023-08-21 by github-actions[bot]