Ultimaker/Cura #16240: Skirt adhesion out of build plate

dakalamin, 2023-07-21:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Creality Ender 3s1

### Name abnormal settings

Build plate adhesion: type = skirt, skirt line count = 2, skirt distance >= 4.56mm. All the rest are default

### Describe model location

Location, scale & rotation - all default

### Describe your model

With skirt distance < 4.56mm (skirt line count -2), brim adheseion type or no adhesion at all slicing seems to be successful

### Add your .zip here ⬇️

STL model and 3MF Cura project: [files.zip](https://github.com/Ultimaker/Cura/files/12130505/files.zip)

[image]

GregValiant, 2023-07-21:
Thanks for the report.
Here is the model sliced with the "Skirt Distance" at 1.
[image]

A new feature of Cura is that the Skirt/Brim can continue off the base and Cura will adjust.  You can see that in the image above.  But SOME of the Skirt/Brim must be on the build plate.  As you can see the skirt curves back in when the model allows it.  Once you adjust the "Skirt Distance" above a certain point then NONE of the skirt will fit on the build plate and slicing fails.

What I see here is expected behavior and there are things you can do to allow the model to slice when it is 218 x 218 on a 220 x 220 build plate.

This isn't a slicing error so I'll remove that label.  I'm further inclined to remove the bug label from this and close it.  Let us know what you think.

dakalamin, 2023-07-21:
Thanks for the fast feedback

I completely agree with Cura's reaction to the skirt missing on the build plate. It's the _unexpected_ error label that confuses me.

The pop-up message doesn't mention the cause of the slicing failure. With my all-plate-wide model it was quite obvious where the problem came from. There was no intention to have a skirt for that print whatsoever - it just happened to be a leftover setting from a previous slicing. But a lucky one with a more complicated model to slice may have some hard time fixing such an issue (and nothing would hint its source as a skirt/brim that's completely out of the printing table is not even displayed).

To make matters worse, the error message explicitly addresses this as a bug and urges to report it. That will definitely result in similar duplicate issues being occasionally published.

I think it'd be better for Cura to be able to recognize such a situation and inform users about it.
[image]
(something like this)

Can't say anything on if it should be a warning or an error, if the slicing should fail or proceed as if the 'Adhesion Type' was 'None' - that's absolutely up to you.

GregValiant, 2023-07-22:
I don't know the inner workings of UltiMaker so some of this is conjecture.
A decision was made in the past to code the Cura Engine in C++ and the Cura front end in a mish-mash of Python with QML dialogs and controls.  I guess they figured it would be easier to maintain one "ambidextrous" front end rather than one for every operating system.
It might be easier to maintain, but not being native to any particular Operating System has drawbacks regarding the controls.  Simple things like informing the user of errors or when settings are in conflict isn't easy to do.  In addition, the UltiMaker Cura team isn't very big and keeping up with the UM printers, firmware, cloud services, Cura, etc.,  is labor intensive.  So little niceties like a full error reporting system fall by the wayside as there are always more pressing needs.  That same small team also seems to be tasked with the Cura Wiki and the web pages.

closed 2023-08-13 by dakalamin