Ultimaker/Cura #16243: Slice varying file size issues, and broken neptune 3 max thumbnail

Gareee, 2023-07-22:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Elegoo Neptune e Max

### Reproduction steps

Sliced a very detailed file the size of the neptune bed. It took 20 minutes or so to slice on a fairly speedy gaming system.

### Actual results

The slice looks fine, and I'm using the neptune thumbnail generator plugin.
What's odd, is every time I save the slice, the file size increases, and none if them display the thumbnail correctly on the printer. 
I suspect either a file size restriction of the thumbnail, or the plugin wasn't considering really complex prints.

### Expected results

The files should save the same size, and the thumbnails should display properly on the printer.

### Add your .zip and screenshots here ⬇️

[image]
Note the different file sizes of the top 5 files...all the same sliced object with no changes...just additional incremental renamed saves.

MariMakes, 2023-07-27:
Hey @Gareee,

Thanks for your report 👍

To me, it doesn't seem like a bug in Ultimaker Cura, but with the community plug-in you are using which is not maintained by us. Are you using the latest version? 
You can download it here: https://github.com/Molodos/ElegooNeptuneThumbnails

I'm not sure how I can help you more than this. 
Would you still like to keep this issue open or do you agree that we can close it?

Gareee, 2023-07-27:
I agree, but I was asked to report it in the Facebook group.That said, having the thumbnail functionality built into cura would avoid the need for the plugin.Btw, the print completed ok, but for some reason brims did not generate on some tree supports, and those eventually failed over the course of the 96 hour print.The quality at 130mm/s looks great though!Sincerely,Gary Tabar Jr.

MariMakes, 2023-07-28:
Hey @Gareee,

If you want to report an issue, I suggest doing it to the maker of the plug-in, that can be done here: https://github.com/Molodos/ElegooNeptuneThumbnails/issues.

If you have any new issues, you can create a new issue for them, so we can track and resolve them separately. 
I'll be closing this issue. Happy Making!

closed 2023-07-28 by MariMakes