Ultimaker/Cura #16248: Slicing fails or hangs. (Failed on larger size, hangs on all others).

chainsawwolf, 2023-07-22:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Anet A8

### Describe model location

Model is centered. Does not slice if rotated either.

### Describe your model

Lithiphane.

### Add your .zip here ⬇️

[litho1_cura.zip](https://github.com/Ultimaker/Cura/files/12136705/litho1_cura.zip)

GregValiant, 2023-07-22:
Thanks for the report and the file.
The model has errors as you can see in this report from[ this repair site.](https://formware.co/OnlineStlRepair)
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 2 Non-manifold edges (?)
--> 2 Inverted faces (?)
--> **1612 Degenerate faces** (?)
--> 2 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 30529ms ------
-> Vertex count changed from 79740 to 80426 (+686)
-> Triangle count changed from 161090 to 160848 (-242)

The repaired file took a LOOONG time to slice.  The Cura team will take a look.  I moved it around, rotated it, and set the Initial Layer Height to 0.23.

[litho1_cura_fixed.zip](https://github.com/Ultimaker/Cura/files/12137069/litho1_cura_fixed.zip)

[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic