Ultimaker/Cura #16249: Unable to open any .stl files at all.

bsaddresss, 2023-07-22:
### Cura Version

5.4.0

### Operating System

Mac Catalina

### Printer

Cr-10S

### Reproduction steps

Cura 5.4.0 beta and fresh installed 5.4.0 will not open any .stl files.  I've tried at least 15 of them, and nothing ever appears on the virtual build-late.  I've tried uninstalls, new installs, restarts, new .stl's from tinkercad, opening .stl's from inside cura, changing default app to Cura... all of the .stl's will still open on prusaslicer.

### Actual results

No Stl will appear on the virtual print bed.

### Expected results

stl should appear on print bed

### Add your .zip and screenshots here ⬇️

na

MariMakes, 2023-07-24:
Hey @bsaddresss,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

Your issue sounds super frustrating 😖 
It reminds me of a similar issue we received earlier this week.

Could it be that our STL reader plug-in is disabled by accident?
If you go to the marketplace, click on the gear, and scroll down until you see STL Reader, you should be able to Enable it again from there.  
[image]

Can you let us know if that is your issue? 
If that's not the case, just let us know we'll take another look. 👍

bsaddresss, 2023-07-24:
Hi, Thank you for your response!

I checked the stl reader and it was enabled.  I disabled it, restarted CURA, re-enabled it, restarted CURA, tried to open a small .stl file (a cheps cube) and 4 minutes have gone by and finally about 4.5 minutes and it opened the file in CURA.

Suggestions are welcome!

 <https://user-images.githubusercontent.com/40423138/255571025-102d8bf6-46c2-497a-93b2-abce5172cb40.png>

dnbernier, 2023-07-28:
@bsaddresss 
I have similar issue.  I can get my STL's to load after I disable the USB Printing plug-in.  Only downside is that printing via computer to usb will not work.

MariMakes, 2023-08-09:
I'll bring it up to the team to see what they can do to improve it.
Since the amounts of reports coming in on the topic is growing.

Fingers crossed 🤞

MariMakes, 2023-08-11:
Quick update from our side 👋

This fix will probably help with with the plug-ins being disabled by accident. 
[Fix required plugins by nallath · Pull Request #891 · Ultimaker/Uranium](https://github.com/Ultimaker/Uranium/pull/891) 
[Move the required plugins list to before it's actually checked by nallath · Pull Request #16304 · Ultimaker/Cura](https://github.com/Ultimaker/Cura/pull/16304)

But they won't help with the workaround that requires you to disable the USB Printing plug-in, that still needs some more research.

MariMakes, 2023-10-11:
The Cura 5.5 Beta has been released 🎉 [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)

We are hoping it resolves some of the issues, but please let us know if that's not the case.

BlondieGayMan, 2023-10-15:
I've had this same problem with every V5 release.
What I have noticed, is that it either takes a long time, or if I launch OR I find that if I launch v4.8 and open the .stl, which appears instantly in 4.8, then, in 5.4.0, I can do file/new and then try to open the .stl... it won't open instantly like in 4.8, but after maybe 30 seconds, then it appears in 5.4.0.

So what I'm wondering, is there any possibility that have several versions installed on the same Window 10 box could have them interfering with each other?

However, to note, I don't have this problem in any version below 5.x.x.  Only in V5.x.x

To be sure, right now, I X'd out of 5.4.0 and launched 5.1 and same issue.  The .stl doesn't load (or takes minutes to appear).

I X out of 5.1, launch 4.8 and it opens the .stl instantly.

Clearly, there's something in 5.x.x causing this problem and I sure wouldn't want to be the coder that has to try and locate the offending library or code or plugin or whatever is causing this.  LOL

Been there, done that....

BlondieGayMan, 2023-10-15:
> The Cura 5.5 Beta has been released 🎉 https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1
> 
> We are hoping it resolves some of the issues, but please let us know if that's not the case.

Sadly, the beta did not work either.
For clarity, the testing I've done is for both just an .stl file (using the same one that works fine in 4.8 and below) as well as the project file.  Neither load right away.

The .stl I tried to open, FINALLY opened after about 4 minutes, so that's faster at least.

Now that something has finally loaded, I can now delete the object, open any other .stl I like without issue.  It just appears instantly.

So it's clear that upon initial launch, there's some library or something missing, that doesn't get loaded right away.  But after waiting for xx minutes and the object finally opens, the library or something is now in memory.  Thus, everything just works fine from then on, UNTIL next fresh launch.

It might be interesting to put break points in the compiler to areas where loading happens and do a before/after list of modules loaded.  Then, after the .stl finally appears, again, do the after list and compare what's missing.  Once found, that could be added to the initial startup tables of modules to load.

Just some thoughts.

darkblaze69, 2024-03-17:
This is still unsolved in 5.6.0. I wait 4 minutes fot stl to load. What it's doing?

BlondieGayMan, 2024-03-17:
Yup!  I don't know why the devs can't find this. On first run of Cura and first load of a .stl, it appears to be stuck in some kind of loop.  Only a loop could cause this long of a delay.
As I said above, once the .stl loads, you can delete it and load another (without closing down Cura, of course) and it just pops up instantly.

I even put my Lab PC to sleep, which some programs have issues with, but Cura plays nicely.  So coming out of sleep, with Cura having already been opened before sleep, it just works perfectly.  STLs open instantly.

Just closing Cura, reopening and loading a .stl triggers the loop.

I'm at the point where I just don't care anymore.  If I need to do some CAD design, I first load Cura and just load any old .stl and let it loop in the background.

By the time I ready to slice what I'm working on, Cura is ready to go.

DJSnCa, 2024-03-30:
I've been experiencing the same issue on arch. running "cura --debug" doesn't seem to show anything other than attempting to read file along with attempting serial communication with a printer followed by changing baud rates

BlondieGayMan, 2024-03-30:
I don't have my printer connected to the computer, but perhaps I could do that to see if by there now being serial comms, that perhaps it loads faster.

IF that is the case, then this is perhaps where devs might want to look in Cura's code.

When Cura itself first loads, it should perhaps check if a printer is connected.
Of even better, perhaps an option in settings where a person can choose to inform that a printer is connected or is not.

If NOT connected, then Cura code should bypass any serial connection attempts.

I may test this next week.  This weekend isn't good for me.  :)

BlondieGayMan, 2024-04-09:
It's interesting that USBPrinting.AutoDetectBaudJob.run seems to be involved.
That makes sense, actually.
So I wonder a couple of things.

For one, I wonder if, at least as a test, adding to the code a JUMP to bypass USBPrinting.AutoDetectBaudJob.run and continue, might reveal that this is the problem.

If so, then the USBPrinting.AutoDetectBaudJob.run code itself needs to be tightened up.

Also, perhaps an option on Cura's Settings to disable that call would resolve it for those of us NOT printing via USB.  ????

I still haven't tested by connecting my printer to USB to see if the auto detect just detects it quickly and moves on.

It's still interesting that after the long delay and the 1st stl finally opens, that closing that .STL and loading a new one is instantaneous.

I wonder if after the first major delay, if
USBPrinting.AutoDetectBaudJob.run
is either bypassed or has cached what it found/didn't find and therefore no longer causes the delay..... hmmmm...

krifpvlic, 2024-04-10:
In the flatpak version of the app, I was able to edit the AutoDetectBaudJob.py file and commented out lines 55-76. This resolved the issue and might work as a temporary remediation for those who don't use the USB printing feature.

gtollini, 2024-05-04:
I fixed this by disabling USB printing altogether (I've never had an USB printer to begin with).
To do that:

- Go to the “Installed” tab of the Marketplace.
- Scroll down to USB Printing and click "disable".
- Restart Cura.

BlondieGayMan, 2024-05-05:
**gtollini**

YOU ARE PURE GENIUS!
I knew that this problem was related to some code loop and I suspected something to do with USB and I had considered connecting my printer to USB just so that Cura could "see" it.

But I would never have thought to go into Marketplace and fine "USB Printing."

So I disabled it as I NEVER use it nor ever would and after restarting Cura, a new .SLT file loaded instantly!

Thank you so much for this.

Perhaps now the Devs/coders can fix this issue for those who may use or not use USB printing.
Perhaps it can be a setting under each printer profile (USB Printing, yes/no) and based on that, just jump over the "USB Printing" code that loops for so long.  Just like disabling it does.

Now that you've confirmed this, the fix would be so easy to do.

I hope that this gets picked up by the Dev team.

Again, A HUGE THANK YOU!

gtollini, 2024-05-06:
I've made a PR with a fix that works for me. It would be interesting for you guys to test it on your machines and report if it works for you as well :smile:. It's [here](https://github.com/Ultimaker/Cura/pull/19025).

As it turns out, it's not scanning each of our printers for USB. It's scanning each of our tty ports (in my case, 32) and trying to make a connection. It'll use a CPU core per port (in my case, 4), 15s per port, twice each. In my case, both in theory and in practice, this totaled 4 minutes between clicking "open" and the file actually opening. 

For some reason unbeknownst to me, this was only enabled on Linux/MacOS. My fix was simply disabling it for everyone, and hopefully a dev with better understanding of the project can add a setting like "Auto Detect USB Printers" or something that we can toggle on or off.

BlondieGayMan, 2024-05-06:
Just for clarity and now I see that I wasn't clear before.  When I said that it seems like it's scanning, I may have said USB ports, but I should have said COMM ports.  That's what you have confirmed.

I agree, I hope that the Devs do 2 things, actually.  One, put an option in settings to configure Enable/Disable like you suggested above.  But upon install of an upgrade apply the user's settings.  On a new install, DISABLE this feature and perhaps have a pop up asking about it.

Secondly, rather than the code doing a full scan of tty/COMM ports to try and find printers, the code should (in Windows for example) read the comm list to see if a printer is there.  I'm not saying that right as I don't code in that area.  What I mean is, that when someone connects a printer to USB, the drivers are installed (again, I use Windows for my examples as I don't do MAC or Linux) so the info is already there.  This should be pulled and not an unnecessary long scanning of all ports.  It really is redundant at the very least.

In a case where someone does have a printer or more connected to USB, once Cura finds it/them I hope that it retains the data so that on next launch, it just launches without trying to find them all again.

Either way, your find and work around is just perfect.  I really appreciate what you've done here.

Thanks again.

gtollini, 2024-05-06:
@BlondieGayMan 

There's something odd there. In theory, my fix wouldn't have a difference on Windows users. All I did was switch a ```Platform.IsWindows()``` (which would be ```True``` for you) for a hardcoded ```True```. Can you build from source from my branch and test it on your system?

It could also be an implementation failure on ```IsWindows()```, I'll investigate it a bit. 

Also, if you don't mind sharing this information, which version of Windows are you running?

BlondieGayMan, 2024-05-11:
I haven't had time (or motivation, to be honest) to test this, but if one does have the printer connected to USB and powered on so the operating system "sees" it, does Cura, if USB printed left enabled, still do the xx minutes thing?
Or since the print is now visible, does Cura "see" it immediately and then just loads the .stl file quickly??

Just curious.

I joke about not having time to test.  It's more that in order to connect the printer, I have to dig out the computer to connect the printer to a USB port at the back.

gtollini, 2024-05-11:
I don't have an USB printer to test this out, but I'd guess it would still take a long time to open files. 
The code tests all tty ports for baud rate - independently of it connects, I think.

BlondieGayMan, 2024-05-12:
Ok.  If I can find some time tomorrow and remember, I'll connect the printer and test.  I'll have to also remember to enable USB Printing.  If I get this done, I'll report back the results here.

What I'm hoping, is that the coders had the sense to look first for an active port and if something found, then test for baud rate.

kafufafu, 2024-06-05:
Could this have had any relation with the following? #19186

gtollini, 2024-06-11:
> Could this have had any relation with the following? #19186

Yes, it's the same root cause.

sneakythr0ws, 2024-07-17:
I have same issue. Model doesn't appear before USBPrinting.AutoDetectBaudJob finishes. 
Disabling usb printing pliugin helped

GregValiant, 2024-07-21:
I keep seeing this thread at the top of the page.  There may be some movement on this as it has come up in other threads.

USB Printing is enabled by default as some older UltiMaker printers require it.  For everyone else, it's a non-maintained legacy feature that not many use (or should use).

The "won't load files on startup" problem seems to be a lot worse when a user has other USB devices plugged into their computer and those devices are configured as serial devices.  Headphones and some other accessories are known issues as Cura keeps polling the device and sending "M105" and the device keeps sending back "What?" and that goes on and on.

I'm in favor of adding a setting to the Extensions menu that will either jump to USB Printing in the MarketPlace where the user could turn it on, or better, just be a switch that would toggle it on and off.  It's installed, loaded, and enabled by default and I see no reason to have to go to a remote website to disable it or enable it.

horga83, 2024-08-05:
Still broken under both X and Wayland on latest Arch and Cura 5.8.0 running under GNOME 46
USB printing disabled and Cura still unusable

horga83, 2024-08-05:
My apologies this may have been a bad STL for those wanting to test it is this project.
https://www.thingiverse.com/thing:3842871/files
The smaller STL opens the larger one does not.

nebomuk, 2024-09-16:
> I keep seeing this thread at the top of the page. There may be some movement on this as it has come up in other threads.
>
> USB Printing is enabled by default as some older UltiMaker printers require [...]

I can reproduce this issue exactly. Download the latest Cura.AppImage (on linux), plug in a USB headphone and it will take a few minutes to load the simplest stl files. 
The cura log is full of entries regarding the USB printing plugin. 
When I disable the USB printing plugin, .stl files just load just instantly.

BlondieGayMan, 2024-09-16:
So here's something.  Due to Microsoft's BS with restoring Win10 images just failing, I wasn't able to restore my LabPC to an SSD.
So, ok... bite the bullet and just do a fresh install of everything.

So on this fresh install, after all updates done and my own config/setup, I downloaded and installed the latest version, 5.8.1.

Without doing anything to it and it couldn't pick up any old settings since this is a fresh install of Windows, I grabbed one of my bigger .STL files and it appeared instantly.

I didn't have to disable anything and yes, the things plugged into the USB ports has not changed.  This includes a 10-port USB hub that I use mostly for Arduino and ESP32 boards.

So it seems to be fine here... at least, for now.

GregValiant, 2024-09-18:
@horga83 
I'm on a Windows 10 system.

There is something wonky about the larger of the models you posted.  It's kind of big for most printers.  This is in MS 3D Builder and Cura shows the same numbers:
[image]

In Cura, at a scale of 0.10% it is 22mm tall.  That makes a lot more sense than 22 meters tall.
[image]

GregValiant, 2024-09-19:
That was just the Z.
In reality (I'm not sure whose) the printer would need to be be 150m x Y100 x 30m (500ft x 325ft x 100ft).  You could print a full size Boeing 747.

In regards to the orientation of that part - it's my printer and I can print however I want.
(The smaller part came in at a reasonable size so I didn't fool with it.)

supertin, 2025-01-16:
I just downloaded 5.9 on a cleanish (never had cura or anything 3D print related) install of Mint. Got this same issue of silently not loading the STL. The log was full of attempts to open non-existent serial ports over and over.
And wouldn't you know it... Disabling the USB printing plugin makes it load.

Now the problem. The SD card reader on my printer is terribly unreliable, so I have been using USB direct printing. So for me, disabling USB printing isn't going to be a real option. So I'm going to have to wedge myself into a space near enough to the printer to set up the job with the printer attached by the sounds of things. Not fun.

Also, how is this still an issue 18 months later? How many versions have passed by now with this bug present?

dzek69, 2025-01-16:
I'm on Garuda Linux right now, on version 5.9, both from AUR and AppImage I have the same issue.

Previously I was on KDE Neon and using AppImage 5.6 version it was all fine (I did not touch any default plugin settings)

buzztiaan, 2025-02-11:
Just confirmed, -all- appimage versions i have on my debian stable install broke very recently. 

Opening -any- file (even presliced gcode) would fail without any message.

Disabling the never used 'USB printing' plugin solved it. 

Would be nice if 'plugins' was just under 'settings' or 'configure' , instead of 'marketplace'

idlesign, 2025-03-16:
Cura 5.9.1 appImage
Ubuntu 24.04.1 LTS

Can confirm
https://github.com/Ultimaker/Cura/issues/16249#issuecomment-2651612879

Disabled USB printing fixed the issue.

0xjairo, 2025-04-27:
Can also confirm that disabling the plugin "USB Printing" solved it

Ubuntu 22.04.5 LTS
Cura 5.10.0

GregValiant, 2025-04-27:
I think in the next Cura release "USB Printing" will be "disabled" by default.
The reason it is still included is that there are a lot of "UM Original" printers still working out there and they require USB Printing.

rrendec, 2025-09-01:
First of all, many thanks to whomever figured out the problem was the USB printing plugin. This was driving me nuts and I reinstalled and reconfigured Cura multiple times, thinking that it was some configuration or library compatibility problem.

For those who are affected by this issue and still want to use USB printing, I found a much better workaround. The answer is in `USBPrinterOutputDeviceManager.py`, and it looks like the USB printing plugin already has support for filtering the serial port devices that are scanned. See lines 132-139.

For me (on Fedora Linux) most of the time is spent scanning ttyS0 through ttyS31, which are not attached to any real devices. My printer uses a ch341 chip internally for the USB interface, and on Linux it's exposed as ttyUSBx (typically ttyUSB0, if I don't have another device already attached). So, for me, starting Cura like this solves the problem:
```
export CURA_DEVICENAMES=USB
cura
```

It still takes a bit of time to probe the various baud rates until it finds the right one, but it's just a few seconds, so it doesn't really matter. And if the printer is not connected, ttyUSB0 is not there, so Cura doesn't spend any time at all scanning.

GregValiant, 2025-09-01:
Thanks for posting that.
This is actually a relatively new problem.  Something changed somewhere else and it affects the way USB Printing interacts with the computer hardware.  A real problem occurs when there are USB devices connected and those devices are internally configured as "serial devices".  Cura assumes it is talking to a printer and sends "M105" to elicit a response.  The earbuds keep going "What!!" and then things go in a circle.
It doesn't always happen, but certain types of USB ports in conjunction with certain operating system installations do show the problem.  It happens on MAC's, Windows, and Linux, but not all of them and not all the time.  Just about impossible to fix without having access to one of every type of printer and computer combination.

dzek69, 2025-09-01:
@horga83 this is exactly what will happen in next version:

https://ultimaker.com/learn/ultimaker-cura-5-10-stable-release-notes/

dzek69, 2025-09-08:
Yes, the usb printing will be disabled by default in 5.11