Ultimaker/Cura #16251: Jordan slicing bug

Warbud666, 2023-07-23:
### Cura Version

5.3.1

### Operating System

windows 11

### Printer

kobra

### Describe your model

Nike shoes

### Add your .zip here ⬇️

[jordan.zip](https://github.com/Ultimaker/Cura/files/12138036/jordan.zip)

GregValiant, 2023-07-24:
Thanks for the report and the file.
I tried a bunch of different things and I can't get it to slice either.
I'll mark this as a duplicate and the Cura team will take a look.

This is sliced in 4.13.1.  Your setup makes it a 65 hour print.  If it was a Chuck Taylor Converse Allstar it would be much faster.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic