Ultimaker/Cura #16252: Adaptive layer under extrusion

Tomtomburke, 2023-07-23:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ender 3 V2 profile

### Reproduction steps

Print stepper motor mount for ender 3 v2 in PETG.
Adaptive layer change produces smaller layers near the top and bottom of holes in the vertical orientation to get a better resolution

### Actual results

Noticed whenever a hole is printed in the vertical orientation the layer height changes at the top and bottom of the holes (makes sense to increase resolution).
However, along these layer changes, I noticed two things, either over extrusion and it creates a bulge/ripple/strip very similar to Z banding (spent a bit troubleshooting my z rod before realizing this setting was checked off). Or you get under extrusion creating visual air pockets where there's no material.
I think tuning the flow rate in these layer changes would resolve this issue. 

### Expected results

Increase resolution to get better overhang results, shorten the print time, and produce a consistent finish across parts. 

### Add your .zip and screenshots here ⬇️

[adaptive_layers.zip](https://github.com/Ultimaker/Cura/files/12140099/adaptive_layers.zip)

GregValiant, 2023-07-24:
Thanks for the report and the files.
I analyzed the Gcode file in MS Excel and all the extrusions are correct.  The flow is 100% given the layer heights of .16 and .20 and the volume of each extrusion results in line widths between .31 minimum and a couple of inner-walls at .65 which is the maximum.

This is your gcode on my Ender 3 Pro.  I changed the temperatures since I had PLA loaded in the machine.  I print a lot of PETG and I would expect these same results.
[image]

The print does not show any defect like we see in this image of your print.
[image]

[image]

I do see a very small misalignment between the .16 layer areas and the .20 layer areas on the ends of those narrow walls.  I don't see any extrusion problems though.  I think the .16 layers were just a tad more accurate at the extrusion starts and ends.

I was going to remove the bug label from this but I've decided to leave it on and the Cura team will take a look.
From what I see in the print comparison is that you had a blip in the filament or in your hot end.  I can't duplicate the problem even using your own gcode.

Tomtomburke, 2023-07-24:
Thanks for looking into this! It is possible it's related to my
machine/filiment. I printed that same part a few times and along that same
layer parts seemed to be under extruded (not the same spot but same layer
height).
My assumption was it's going from 0.16 to 0.2 layer height. And that layer
change may not be extruding more filiment which would be needed to fill the
gap.
I didn't look at the gcode to verify this yet, as im not very familiar with
the structure of gcode just yet.

I do appreciate taking time to try and reproduce the issue, fantastic
support from you guys!
I figured since it's experimental it might be worth reporting incase others
have similar issues.

Thanks!
-Thomas

GregValiant, 2023-07-24:
I've been thinking on this and what I think is that I'll remove the bug label and close this.  I'll keep an eye out for more problems like this and if it comes up again then this can be revisited.
Adaptive Layers has been in the Experimental section for a long time.  It has had a couple of issues over the last two or three years, but all-in-all it hasn't really been a problem.

closed 2023-07-24 by GregValiant